ericstoneking/42 #3: Citation

acxz, 2019-05-18:
This is a substantial piece of software that should be citable. It would be great if a citation along with a bibtex entry can be provided for this project.

ericstoneking, 2019-05-20:
I didn't have the patience to do it the helpful link's way, but here's a stab at a bibtex entry:

\bibitem{42}
Stoneking, Eric, {\it 42: A General-Purpose Spacecraft Simulation}, NASA Software Designation GSC-16720-1, 2010-2019.  https://sourceforge.net/projects/fortytwospacecraftsimulation, https://github.com/ericstoneking/42.

Check me on the URL formatting (my LaTeX has apparently been kneecapped by a recent system migration).  I'll put this into the License folder in my next drop.

acxz, 2019-05-20:
The formatting is just fine!
I was thinking more of something like this tho for a bibtex entry:
```
@misc{Stoneking10,
  author          = {Eric Stoneking},
  title           = {{42: A General-Purpose Spacecraft Simulation}},
  year            = {2010--2019},
  howpublished    = {NASA Software Designation GSC-16720-1},
  url             = {https://sourceforge.net/projects/fortytwospacecraftsimulation, https://github.com/ericstoneking/42}
}
```
*Not sure if the howpublished field is correctly applicable.

So that people can just zap it into their .bib file and have their bibliography style generate the format they need for the citation.

But yeah I think having both the default citation and the bibtex entry is a good idea.
```
Stoneking, Eric, 42: A General-Purpose Spacecraft Simulation, NASA Software Designation GSC-16720-1, 2010-2019. https://sourceforge.net/projects/fortytwospacecraftsimulation, https://github.com/ericstoneking/42.
```

acxz, 2019-12-20:
Closing this issue since the [HowToCite.txt](https://github.com/ericstoneking/42/blob/master/License/HowToCite.txt) file has been added with the relevant citation information.

closed 2019-12-20 by acxz