ericstoneking/42 #4: Confusing READMEs

acxz, 2019-05-18:
Having two READMEs is confusing. Is it possible to merge the two?

ericstoneking, 2019-05-20:
If you've only found two, you have more looking around to do :-)  FWIW, the README.md is apparently autogenerated by github from the (main) Readme.txt.  The txt version is primary, and is not to be deleted.  (To minimize dependencies, I standardize on ASCII text.  I don't mind if git processes that, but it doesn't get to supplant the original, portable version.)  Make your edits to the txt version, and I'll accept them.

acxz, 2019-05-20:
Markdown (.md) is also an ASCII text format and very readable. The added benefit is that markdown can be interpreted as formatted text with sections, subsections, codeblocks, bullet points, etc. straight from ascii text. I really believe having a markdown Readme would be the best option.
 You can see my branches with [markdown](https://github.com/acxz/42/tree/master) and with [txt](https://github.com/acxz/42/tree/readme_fix) to notice the differences with the exact same file contents but just a different file extension. If you would still like to stick with the txt version, I'll go ahead and create a PR with that.

ericstoneking, 2019-05-21:
Why not both?  I don't mind the md formatting at all.  Knock yourself out.  All I'm saying is that there should be one readme with a txt extension, for portability.

acxz, 2019-05-21:
okay then, I added a copy of README.md and renamed it README.txt

closed 2019-05-24 by ericstoneking