ericstoneking/42 #11: Release Version

acxz, 2019-06-16:
It would be beneficial to have a [release](https://github.com/ericstoneking/42/releases) for 42, so that users can download a release version instead of the latest master branch, which may or may not be stable.

ericstoneking, 2019-07-17:
Sorry for the slow response.  Crunch time at the day job.

I'm open to the idea, but not sure I can live up to the expectations it may raise.  42 is my workhorse simulation tool.  As such, its growth is rather organic.  Every once in a while, I wrap it up and put it up on sourceforge (as a zip file) and push it here to github in a fire-and-forget way.  I don't want to get too involved maintaining it on github because I already maintain it on my internal svn repo.  Any bug reports or feature requests, either from here or sourceforge, get addressed first in my local environment, and then get pushed out on my next upload whenever that happens.

I have taken to naming the sourceforge drops with the upload date, which I reckon serves as a version number of sorts.  If releases here on github work the same way (i.e. here's what Eric put up on this day), then it sounds harmless enough.  What I don't want to do is manage variations between, say, v1.5 and v1.6.  Only the latest drop matters, and I reckon that's what I get by only having a master branch.  I'm not a software guy by trade, though, so I'm open to hearing how releases will make my life better.  Extoll away :-)

acxz, 2019-07-20:
I see. If the latest drop is the only version/commit that matters, then I understand not creating any releases. Users can always just download the latest master branch source code.

closed 2019-10-09 by ericstoneking