ericstoneking/42 #12: How can I show all the spacecrafts at the sametime

wangzhanwei666, 2019-07-17:
#
**************************  Reference Orbits  **************************
2                               !  Number of Reference Orbits
TRUE   Orb_LEO.txt              !  Input file name for Orb 0
TRUE   Orb_LEO1.txt             !  Input file name for Orb 1
*****************************  Spacecraft  *****************************
2                               !  Number of Spacecraft
TRUE  0 SC_Simple.txt          !  Existence, RefOrb, Input file for SC 0
TRUE  1 SC_Simple1.txt          !  Existence, RefOrb, Input file for SC 1

This is my set, but I can show only one spacecraft at the map.

wangzhanwei666, 2019-07-17:
My purpose is to simulate the OneWeb constellation.

ericstoneking, 2019-07-17:
The map was set up to only show the spacecraft that was the "Host" of the point of view (POV).  I just pushed an update so that the map shows all orbits around the same world as the POV host.  Run "42 Demo" and cycle through the hosts to see it in action.

closed 2019-10-09 by ericstoneking