ericstoneking/42 #15: Planetary propagation of Mars seems inconsistent with other simulations

kencenter, 2019-10-25:
I have created several 42 scenarios involving S/C in orbit around Mars. Those scenarios have time epochs in 2020. 

To verify the locations of orbits over features on the Martian surface I created an identical scenario in STK, which did not agree with 42. I noticed that in addition to the orbits not matching up, Mars' surface solar illumination condition is also in disagreement (wrong face of the planet is facing toward the Sun). This is more likely the root issue than the orbits themselves. 

I have struggled for some time trying to find problems in my 42 input files, but so far have not unearthed anything. Thinking the problem could conceivably be in STK, I used JPL's Solar System Simulator (https://space.jpl.nasa.gov/) as a tie breaker. It agreed with STK...

So I am hoping to get some guidance on whether I have missed a critical detail or if the propagation of Mars planetary state is indeed incorrect. I plan to set up 42 scenarios for some of the other planets to see if inconsistencies appear there as well. 

Oh... For the record, Earth is fine. Any 42 scenario I create involving Earth is completely consistent with STK.

ericstoneking, 2019-10-25:
Hi Ken,

My fault entirely.  I hope it hasn't caused you too much trouble.  I don't properly initialize the rotation state of the planets and moons, with the exception of Earth and Luna.  I didn't have a reference for these, so I just punted and assumed a PriMerAng (prime meridian angle, the analog of GMST) of zero at the J2000 epoch (the origin of AbsTime).

The code changes to fix this are easy, but I don't have the data to fill in the PriMerAng at J2000.  If you happen to have these, please send them my way.  Otherwise, I'll see what the internet can provide.

Regards,
-Eric

kencenter, 2019-10-25:
Hi Eric,

No this issue has not caused me that much trouble. I just always suspect that it's me and didn't want to reach out until I did my due diligence. I'm just pleased to know that the issue is acknowledged and understood. I asked internally (within my organization) about whether anyone knows of a source for the Prime Meridian angle of the planets at the J2000 Epoch. One answer back was that the SPICE toolkit (open source) probably has them. It would be wonderful to get this fixed if possible. We're hoping to leverage 42 for demonstrations of collaborative autonomous robotic teams operating in the Mars environment on a JPL STTR.

kencenter, 2019-10-25:
Eric,

The Wo column of the table at this link (https://www2.mps.mpg.de/homes/fraenz/systems/systems3art/node18.html) may have the values you need. If you compare the value of the Earth row to the value you use to initialize Earth in 42, I'd expect the other planet's values to be good as well.

-Ken

ericstoneking, 2019-10-27:
Hi Ken,

Thanks for the reference.  I've made the updates so all the planets should have plausible rotation phases now.  You may want to tweak Mars's PriMerAngJ2000 and/or rotation rate to match your other sources.  See 42init.c, lines 2378 and 2384.  Let me know if you have any problems.

Regards,
-Eric

kencenter, 2019-10-27:
Eric,

I synched up and am happy to report that my 42 Mars scenarios now look correct (in terms of solar lighting and S/C ground tracks) - good agreement with STK... Thanks so much for the rapid fix! 

-Ken

closed 2019-10-27 by kencenter