ericstoneking/42 #20: Simulation Pauses if the front end visualization view is adjusted

crgnam, 2020-02-04:
I'm not sure if this is expected behavior or not, but I'm currently in the process of hooking 42 to our flight software implementation (running on our hardware).  I noticed however that whenever I adjust the view in the CAM window, that the simulation appears to pause.  I've tried setting Inp_Sim.txt to both REAL and EXTERNAL, however I've had no luck there.  Anytime the mouse left clicks in the CAM window, the simulation pauses.

I'm currently running all of this on WSL (Ubuntu on Windows 10).  I've noticed no other issues since getting started.

If this truly is expected behavior, we can make due without adjusting the visualization.  It would just be extremely useful for an upcoming live demo we have to do with all of our hardware in the loop.

Please let me know if there is anything I'm able to do!

crgnam, 2020-02-04:
I was able to figure it out.  I used the Rx/Tx example as a basis for the communication, splitting the simulation and front end into different processes.  It now behaves just as I needed!  I apologize for posting.  

Thanks for the great software!

closed 2020-02-04 by crgnam

ericstoneking, 2020-02-05:
Hi Chris,

Questions you answer yourself are my favorite kind :-)

The behavior you opened with is expected behavior.  It's a feature when everything is embedded and running as one process (which is where I spend most of my time).  You're not the first to be annoyed by it in a more real-time environment.  Anyone who wants to alter it could start by searching for "PausedByMouse" in 42GlutGui.c.  Looks like you found an equally good, simple solution.

Regards,
-Eric