ericstoneking/42 #21: Use of EXTERNAL TimeMode option

kencenter, 2020-02-05:
Hi Eric,

I'm reviewing docs and code and having a difficult time understanding how the EXTERNAL_TIME option is supposed to work. I'm hoping to synchronize 42 with another simulation (which cannot be slaved by a time master), so I'm hoping 42 can be configured to slave to that other sim, hopefully by consuming a "dt tick" message or some similar mechanism. Can you point me in the right direction on understanding how the EXTERNAL_TIME option currently works? If I change my Inp_Sim.txt file to use that option (instead of REAL_TIME or FAST_TIME) I don't see 42 behaving any differently than if I use the REAL_TIME option.

Best,
-Ken Center

kencenter, 2020-02-05:
Eric,
I think I just discovered something, which is that EXTERNAL simply uses current time (as determined by the local machine's clock), as opposed to the initial time specified in the input file. 

Given that, I'm still wondering if you have a mechanism in place to support slaving 42 to some other simulator's concept of time. If not, I can extend my local code copy to accomplish this, but did not want to duplicate effort if there was already a solution in place that I'm just not seeing.

Thanks again,
-Ken Center

ericstoneking, 2020-02-06:
Hi Ken,

I can see two approaches that should be doable:
1)  Invent your own TimeMode, using the cases in 42exec.c:AdvanceTime as starting points.  For example, if you can replace the call to RealSystemTime with something that reads your other simulator's time, that should do it.  (See 42nos3.c for another example.)
2)  If you haven't already, familiarize yourself with the TxRx example (see Docs/TxRx Example.txt).  Replace the Tx end with your other simulator, and have it send time over the socket as a text message in the form:
TIME 2016-081-12:01:39.200000000
[EOF]

Regards,
-Eric

closed 2020-05-08 by kencenter