ericstoneking/42 #25: 42 Not Able to Run Due to Syntax Error

evanpw3, 2020-04-01:
Hi there!

I am extremely new to coding in any sort of language, but I was asked to look into getting this program to work for my design team at school. I went through the steps to download msys and also Ubuntu and attempted to execute 42 within both of them. I kept receiving the same error "line 3950: syntax error: unexpected end of file" so it would not run. I was wondering why this was as when I opened up the code to take a look at line 3950, it did not exist? It only goes up to line 3949.

Thank you for your time,
Evan Warner

ericstoneking, 2020-04-02:
Hi Evan,

Welcome to the wonderful world of programming!  I can try to get you past this bug, but you should probably line up some local help, because there are sure to be more to come.

To see what your problem is, I need more info.  In your terminal window, enter these:
```
make clean
make $1 2>&1 | tee makelog.txt
```
That should capture all the results of make into the file makelog.txt.  Then:
```
42 $1 2>&1 | tee runlog.txt
```
Send me makelog.txt and runlog.txt.

A little explanation:  42 is composed of many files.  The ones that end in .c and .h are called _source files_.  The make utility uses the Makefile as a recipe to turn all those source files into _object files_.  This step in the process is called _compilation_.  Then make _links_ all the object files together to create the _executable_, which is the program that you actually run.  It isn't clear from your initial description which part of the process failed, or which file was being referred to.  The log files that you send me will help pin this down.

evanpw3, 2020-04-02:
Hi Eric!

Thank you so much for the guidance! I was able to make the makelog.txt and
the runlog.txt, but for the runlog.txt, it just gave me the prompt "42:
command not found".

I'll be attaching them to this email.

Thank you for your time,
Evan Warner

gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42main.c -o ./Object/42main.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42exec.c -o ./Object/42exec.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42actuators.c -o ./Object/42actuators.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42cmd.c -o ./Object/42cmd.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42dynamics.c -o ./Object/42dynamics.o  
./Source/42dynamics.c: In function ‘FindInertiaTrq’:
./Source/42dynamics.c:1347:19: warning: variable ‘Nf’ set but not used [-Wunused-but-set-variable]
       long Ib,i,j,Nf;
                   ^
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42environs.c -o ./Object/42environs.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42ephem.c -o ./Object/42ephem.o  
./Source/42ephem.c: In function ‘Ephemerides’:
./Source/42ephem.c:514:35: warning: variable ‘EarthMoonBaryVelH’ set but not used [-Wunused-but-set-variable]
       double EarthMoonBaryPosH[3],EarthMoonBaryVelH[3];
                                   ^
./Source/42ephem.c:514:14: warning: variable ‘EarthMoonBaryPosH’ set but not used [-Wunused-but-set-variable]
       double EarthMoonBaryPosH[3],EarthMoonBaryVelH[3];
              ^
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42fsw.c -o ./Object/42fsw.o  
./Source/42fsw.c: In function ‘CmgFSW’:
./Source/42fsw.c:1438:14: warning: variable ‘Gain’ set but not used [-Wunused-but-set-variable]
       double Gain;
              ^
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42init.c -o ./Object/42init.o  
./Source/42init.c: In function ‘LoadDE430’:
./Source/42init.c:3436:35: warning: variable ‘EarthMoonBaryVelH’ set but not used [-Wunused-but-set-variable]
       double EarthMoonBaryPosH[3],EarthMoonBaryVelH[3];
                                   ^
./Source/42init.c:3436:14: warning: variable ‘EarthMoonBaryPosH’ set but not used [-Wunused-but-set-variable]
       double EarthMoonBaryPosH[3],EarthMoonBaryVelH[3];
              ^
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42ipc.c -o ./Object/42ipc.o 
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42perturb.c -o ./Object/42perturb.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42report.c -o ./Object/42report.o  
./Source/42report.c: In function ‘Report’:
./Source/42report.c:129:20: warning: variable ‘ProjAreaFile’ set but not used [-Wunused-but-set-variable]
       static FILE *ProjAreaFile;
                    ^
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42sensors.c -o ./Object/42sensors.o  
gcc -Wall -Wshadow -Wno-deprecated -g   -I ./Kit//include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/  -O0  -D _USE_GUI_ -D _USE_SHADERS_ -D _USE_SYSTEM_TIME_ -D _ENABLE_SOCKETS_   -D _USE_GSFC_WATERMARK_    -c ./Source/42nos3.c -o ./Object/42nos3.o  
./Source/42nos3.c: In function ‘ReadNos3InpFile’:
[... cut ...] 2sensors.o ./Object/42nos3.o ./Object/42GlutGui.o ./Object/glkit.o    ./Object/dcmkit.o ./Object/envkit.o ./Object/fswkit.o ./Object/geomkit.o ./Object/iokit.o ./Object/mathkit.o ./Object/nrlmsise00kit.o ./Object/msis86kit.o ./Object/orbkit.o ./Object/radbeltkit.o ./Object/sigkit.o ./Object/sphkit.o ./Object/timekit.o ./Object/AcApp.o   ./Object/SimWriteToFile.o ./Object/SimWriteToSocket.o ./Object/SimReadFromFile.o ./Object/SimReadFromSocket.o  -lglut -lGLU -lGL -ldl -lm 

42: command not found

ericstoneking, 2020-04-02:
Hi Evan,

Okay, the compiler warnings all look harmless.  So far, so good.

"Command not found" is a classic.  Try this instead:
```
./42 $1 2>&1 | tee runlog.txt
```
The ./ tells the operating system that the command you want (42) is located in the current directory (.).  If you get tired of typing the ./, you can add the current directory to your path.  But that's beyond the scope of this bug hunt.  For now, use the ./, and send me the results.

evanpw3, 2020-04-02:
Hi Eric,

Okay awesome! That looked like it did something.

Thank you for your time,
Evan Warner

0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000
freeglut (./42): failed to open display ''
Initializing GLUT

ericstoneking, 2020-04-02:
Hi Evan,

Are you on msys or ubuntu?  If ubuntu, then study the part of "Docs/Install Ubuntu on Win10.txt" about setting the DISPLAY, and try again with this modification:
```
DISPLAY=:1 ./42 $1 2>&1 | tee runlog.txt
```
Substitute the correct display number as appropriate.  If it works, send me the resulting runlog.txt.

If that doesn't work, then I'm out of ideas.  freeglut is the graphics "engine", so the problem is somewhere with that getting along with your system.  One thing that helps sometimes is to make sure that your video drivers are up to date, but that is a more common issue on linux systems.

If you can't resolve this, and you're willing to use 42 without the graphics front end, then you still have a fallback to try.  Edit your Makefile, and make these lines:
```
GUIFLAG = -D _USE_GUI_
#GUIFLAG =
```
look like this:
```
#GUIFLAG = -D _USE_GUI_
GUIFLAG =
```
Graphics are useful and cool, but optional.

evanpw3, 2020-04-02:
Hi Eric!

It seems to have worked!! Here is the resulting runlog.txt.

Also is anything supposed to pop up in the "42 cam" window? Because the "42
Map", "42 Unit Sphere Viewer", and "42 Orrery" windows popped up and are
displaying stuff, but the "42 Cam" one is not displaying anything.

Thank you for your time,
Evan Warner

0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000

evanpw3, 2020-04-02:
Hi Eric,

Also I did not realize the runlog.txt would update when I exited out of the
resulting windows. Here is the updated file.

Thank you for your time,
Evan Warner

0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000
Initializing GLUT
Initializing Cam Window
Loading Cam Shaders
Loading Cam Textures
Loading 3D Noise
Loading Cam Lists
Done Initializing Cam Window
    42 Case ./InOut/ is   0% Complete at Time =        0.000

closed 2020-09-11 by ericstoneking