ericstoneking/42 #28: Tx/Rx example crashing

twaters, 2020-04-27:
We have setup a simple data server to push PVT to 42 for visualization of our simulation, but find 42 has been crashing after a few seconds.  I went back and ran the Tx/Rx example,, but the rx instance also segaults after a few seconds.  Unfortunately I haven't been able to generate a core file to try and debug. 

What conditions are you aware of that cause the Rx instance to default?  I'm currently running off develop.

ericstoneking, 2020-04-28:
I think you've found a bug.  If it is what I think it is, InitAC is being called twice, once from 42init.c and once from 42fsw.c, and so some parameters get zeroed.  This is not the intended behavior.  I just pushed the fix.  See if that addresses your situation.

twaters, 2020-04-28:
That has definitely helped, but there must be a second issue.  I was able to spin up a debugger and catch the crash.  I didn't dig in very far, but at least got a stack trace:

<some library stuff here>
glkit.c: line 37
42GlutGui.c: line 754
42GlutGui.c: line 1566
glutMainLoop() (no line number)
42GlutGui.c: line 6845
42exec.c: line 374
42main.c: line 41

I know this is only partial info - I'll see if I can get some additional time tomorrow to investigate further.

closed 2020-09-11 by ericstoneking