ericstoneking/42 #45: 42 Cam graphics choppy using VMware with an Ubuntu guest

chupalt, 2020-10-07:
A recent commit has made the graphics choppy in the 42 cam when running in VMware with an Ubuntu guest.  This problem seems to be recently introduced.  An early September clone of the repo did not exhibit this behavior.  Seeing the following error in the dmesg log.  

[drm:vmw_stdu_primary_plane_atomic_update [vmwgfx]] *ERROR* Failed to update STDU.

We will continue to review recent commits to see if we see an obvious error and will report back if we do.

chupalt, 2020-10-07:
It seems like the issue was introduced in this commit.  f84857f016429320324bc00f8e9ab3c34a03fdae

chupalt, 2020-10-07:
Commenting out line 3688 in 42GlutGui.c fixes the issue on my system.

[image]

ThatcherC, 2020-10-28:
I'm also seeing weird graphics on Ubuntu 20.04 native - the Cam view seems to go forward smoothly for a second, then kind of jump back a bit, then go smoothly for another second, then jump back, over and over. I could try to share a video if that would help.

@chupalt's fix works for me too. It also has the nice side-effect of solving a different weird graphics bug I've been having where the Cam and Orrery windows resize uncontrollably whenever I try to drag the window around. I had gotten around that issue by commenting out line 6163 in 42GlutGui.c, but that removes the ability to resize the cam window at all:

https://github.com/ericstoneking/42/blob/e6862f6ec3895f2fed5b0489eea319d6da817e44/Source/42GlutGui.c#L6162-L6164

I don't know enough about the graphics code here, but I'll keep using @chupalt's fix for a while and see how it goes.

vladtcvs, 2020-11-09:
Also have same problem on host system, without vmware #52

ericstoneking, 2021-08-15:
I don't know why this works, but it does.  I had no issues on OSX or msys, and my experience on Ubuntu isn't exactly what y'all are seeing, but it's clear to me that Ubuntu/xterm/whatever are happier without that call than with it.  (And OSX and msys don't miss it when it's gone...)  Look for a new code drop soon.