ericstoneking/42 #47: Running 42 in background with no GUI causes IPC to NOT be loaded

PeteBlanchard, 2020-10-12:
Was doing a test running 42 in the background (NOS3 build) and ran into a weird issue, was wondering if you might have an idea as to what is happening...
If I start the simulation with the GUI set to false via a command like this, it works fine:

    cd /opt/42
    ./42 <sim folder>

However, if I then try to run in the background (and save the log) it stops after it reads the Inp_Cmd.txt, but fails to process the IPS or step through the simulation, it just hangs:

    ./42 <sim folder> | tee /tmp/42.log

Produces:

    Material material_3 is not in Matl structure
    Setting material to default
    0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
    Reached CmdScript EOF at Time = 0.000000

Looking at the 42exec.c, it seems like it should be fine...after processing the command interpreter, it checks whether sockets are enabled ('_ENABLE_SOCKETS_'; they are, but don't see the 'server listening'), then it checks '_USE_GUI_' (it is), but the 'FALSE' should have set 'GLEnable' to false, so it should drop into the 'SimStep' processing.

Thanks in advance.

PeteBlanchard, 2020-10-12:
update, not just with 'tee'...anything that redirects input seems to cause this...e.g.

    ./42 <sim folder> 2>&1 > /tmp/42.log

If output is NOT redirected, it works as expected...

ericstoneking, 2020-10-13:
Hi Pete,

I think you're being tripped up by a poorly-documented feature.  42 accepts two command-line arguments:
./42 InOutPath ModelPath

By default, InOutPath = InOut and ModelPath = Model.  You've seen cases where you set InOutPath to whatever you want.  The need for multiple ModelPaths hasn't really ever come up.

So try this:
./42 <sim folder> Model 2>&1 > /tmp/42.log

Regards,
-Eric

ericstoneking, 2020-10-13:
Oops.  Correction to my previous comment:

./42 YourSimCase Model 2>&1 > /tmp/42.log

PeteBlanchard, 2020-10-13:
That seems reasonable, but I get the same result (after changing), stops after executing the command script and before IPC.

ericstoneking, 2020-10-13:
Hi Pete,

So I tried it on my installation (which I should have done in the first place).  These commands all worked fine with or without graphics:

42 2>&1 | tee log.42
42 InOut 2>&1 | tee log.42
42 InOut Model 2>&1 | tee log.42

I don't have any bright ideas about what's causing your issue.  This is about where I would "make clean" and break out the debugger.

Regards,
-Eric

PeteBlanchard, 2020-10-13:
Bizzare, I can't get Visual Studio Code to actually stop at a breakpoint when running...but it appears to start processing the IPC and get's stuck on this line (from Kit/iokit.c, line 159):

          if (bind(init_sockfd,(struct sockaddr *) &Server,sizeof(Server)) < 0) {

because if I cntrl+c (after awhile of being hung) it will print out:

    Error on binding server socket.

But no idea why that is occurring when doing re-direction with no GUI...

Thanks for the help, Eric...guess not much more to do from your end.

PeteBlanchard, 2020-10-22:
One more thing...I forgot to mention that I was starting 42 in the background...

```
./42 <sim> 2>&1 > /tmp/42.log &
```

I am running under NOS3 environment, but I wouldn't think that would be an issue (though it is a slightly older version of 42, not master)

PeteBlanchard, 2021-03-01:
Never figured out what was going on with this...may have been VM issue...did get it working though, your code was fine.

closed 2021-03-01 by PeteBlanchard