ericstoneking/42 #49: Reached CmdScript EOF at Time = 0.000000

timothy, 2020-10-24:
I am getting inconsistent behavior in 42. Some times it will start out with similar messages to below and then recover and some times go on for a long time and then ends with `Reached CmdScript EOF at Time = 0.000000` Any idea what is causing it? 

I see `material_4` is in `stf1_red.obj` and I see other `material_`## variables in other `.obj` files does this mean the obj files are being overwritten or corrupted?

```
...
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_10 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_10 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_10 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_10 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_12 is not in Matl structure
Setting material to default
Material material_11 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_12 is not in Matl structure
Setting material to default
Material material_11 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_12 is not in Matl structure
Setting material to default
Material material_11 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_12 is not in Matl structure
Setting material to default
Material material_11 is not in Matl structure
Setting material to default
Material material_15 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_16 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_16 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_16 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_8 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_16 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_9 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_1 is not in Matl structure
Setting material to default
Material material_4 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_4 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
Material material_4 is not in Matl structure
Setting material to default
Material material_5 is not in Matl structure
Setting material to default
...

Material material_4 is not in Matl structure
Setting material to default
Material material_3 is not in Matl structure
Setting material to default
0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000
```

ericstoneking, 2020-10-26:
Hi Timothy,

stf1_red.obj refers to some materials that don't exist.  Either add them to 42.mtl, or find/create another mtl file (say, stf1_red.mtl) and define them there, then refer to that mtl file instead of 42.mtl.  I don't recognize that model, so I can't say how it got broken.

Regards,
-Eric

timothy, 2020-10-27:
After putting in some fflush in different parts of the code think I found my problem. 

Thank you for your response @ericstoneking!

closed 2020-12-07 by timothy