ericstoneking/42 #51: Questions: Regarding use of "Point" command in Inp_Cmd.txt

PeteBlanchard, 2020-11-04:
Eric,

Sorry to keep bugging you, I was trying to trace through the code myself, but there is quite a bit of it.  When using the "Point" command in the 'Inp_Cmd.txt' file (say a lat/lon on Earth and using the PROTOTYPE_FSW), I had a couple questions:
* how quickly does the satellite turn? (does the 'VectorRampCoastGlide' limit the speed or cause it to taper off?  I see it changing IdealTrq's but ...)
* once pointed at the spot, will the SC continue to point there as it orbits (e.g. try to look through the Earth when on the other side)?  It seems that it would unless something changes the Mode of the command vector
* how do the secondary vectors get populated?  I see they are zeros when calloc'd memory for the structure, but if I set the Primary vector with a "Point" command, I see the secondary vector is used (in calculations) as well, but don't see how they gets set (without a specific command).

Thanks in advance

annamaskolenko, 2022-07-27:
Hello.
Where do you see IdealTrq changing?

PeteBlanchard, 2022-07-27:
Poor choice of words there...I saw IdealTrq set in flight software model initialization...

annamaskolenko, 2022-07-27:
When I am trying to make SC_Simple to point to the Earth using the following command
10.0 Point SC[0].B[0] Primary Vector [0.0 1.0 0.0] at World[0]
nothing happens. Spacecraft still rotates and from CAM window it is visually clear that the SC is not pointing to the Earth.
Nevertheless, wbn output data produced is different from what it was with no pointing command.
I am wondering if you figured out how to use the pointing command properly. I would be grateful if you could help me with it.

ericstoneking, 2022-07-27:
I should point out that World[0] is the Sun.  Earth is World[3].  For the complete list, see 42defines.h

annamaskolenko, 2022-07-27:
Thank you! But still how quickly does the satellite turn? I have pointed its y axis towards World[3], i.e. Earth, but it is still rotating. Does PROTOTYPE_FSW use ideal actuators to execute the command?

ericstoneking, 2022-07-28:
Hi Pete,

For whatever reason, I never saw this issue until yesterday, so sorry for the slightly slow response.

VectorRampCoastGlide's parameters are initialized in 42fsw.c:InitAC.  My copy has a vmax of 0.5 deg/sec.

Yes, PrototypeCtrl will look at the spot you tell it to, whether that spot is visible or not.

Secondary vectors get populated by you.  For the main body, if you don't define a SecVec, the sim should complain and crash.  If it doesn't, I'd need to more about your setup.  For an appendage, a PriVec and the constraints imposed by being an appendage are enough to fully define an attitude, so you don't need a SecVec. 

Regards,
-Eric

ericstoneking, 2022-07-28:
Hi Anna,

Max slew rate is 0.5 deg/sec.  PROTOTYPE_FSW does use ideal actuators.  To monitor those, you may need to write them to an output file (see 42report.c to see how this is done in general) or use a debugger.

So, if it isn't pointing at the Earth, is there anything that it *is* pointing at?  There may be clues there.

Regards,
-Eric

annamaskolenko, 2022-07-28:
Hi
I have the following spacecraft's configuration:

<<<<<<<<<<<<<<<<<  42: Spacecraft Description File   >>>>>>>>>>>>>>>>>
3-U Cubesat                   !  Description
"Cube 3"                      !  Label
GenScSpriteAlpha.ppm          !  Sprite File Name
PROTOTYPE_FSW                       !  Flight Software Identifier
0.2                           !  FSW Sample Time, sec
************************* Orbit Parameters ****************************
ENCKE                         !  Orbit Prop FIXED, EULER_HILL, or ENCKE
CM                            !  Pos of CM or ORIGIN, wrt F
0.0   0.0  0.0 		      ! Pos wrt F
0.0   0.0  0.0 		      ! Vel wrt F
*************************** Initial Attitude ***************************
LAL                           ! Ang Vel wrt [NL], Att [QA] wrt [NLF]
0.0    0.0    0.0             ! Ang Vel (deg/sec)
0.0    0.0    0.0    1.0      ! Quaternion
0.0    0.0    0.0    123  ! Angles (deg) & Euler Sequence
***************************  Dynamics Flags  ***************************
KIN_JOINT                     ! Rotation STEADY, KIN_JOINT, or DYN_JOINT
FALSE                         ! Passive Joint Forces and Torques Enabled
FALSE                         ! Compute Constraint Forces and Torques
REFPT_CM                      ! Mass Props referenced to REFPT_CM or REFPT_JOINT
FALSE                         ! Flex Active
FALSE                         ! Include 2nd Order Flex Terms
2.0                           ! Drag Coefficient
************************************************************************
************************* Body Parameters ******************************
************************************************************************
1                             ! Number of Bodies
================================ Body 0 ================================
3.817                         ! Mass
0.042  0.042  0.007           ! Moments of Inertia (kg-m^2)
0.0  0.0  0.0                 ! Products of Inertia (xy,xz,yz)
0.0  0.0  0.0                 ! Location of mass center, m
0.0  0.0  0.0                 ! Constant Embedded Momentum (Nms)
Cubesat_3U.obj                ! Geometry Input File Name
NONE                          ! Flex File Name
************************************************************************
*************************** Joint Parameters ***************************
************************************************************************
         (Number of Joints is Number of Bodies minus one)
============================== Joint 0 ================================
0 1                           ! Inner, outer body indices
1   231   GIMBAL              ! RotDOF, Seq, GIMBAL or SPHERICAL
0   123                       ! TrnDOF, Seq
FALSE  FALSE  FALSE           ! RotDOF Locked
FALSE  FALSE  FALSE           ! TrnDOF Locked
0.0    0.0    0.0             ! Initial Angles [deg]
0.0    0.0    0.0             ! Initial Rates, deg/sec
0.0    0.0    0.0             ! Initial Displacements [m]
0.0    0.0    0.0             ! Initial Displacement Rates, m/sec
0.0   0.0  0.0  312           ! Bi to Gi Static Angles [deg] & Seq
0.0   0.0  0.0  312           ! Go to Bo Static Angles [deg] & Seq
0.0   0.0  0.0                ! Position wrt inner body origin, m
0.0   0.0  0.0                ! Position wrt outer body origin, m
0.0   0.0  0.0                ! Rot Passive Spring Coefficients (Nm/rad)
0.0   0.0  0.0                ! Rot Passive Damping Coefficients (Nms/rad)
0.0   0.0  0.0                ! Trn Passive Spring Coefficients (N/m)
0.0   0.0  0.0                ! Trn Passive Damping Coefficients (Ns/m)
*************************** Wheel Parameters ***************************
0                             ! Number of wheels
=============================  Wheel 0  ================================
0.0                           ! Initial Momentum, N-m-sec
1.0   0.0   0.0               ! Wheel Axis Components, [X, Y, Z]
0.14   50.0                   ! Max Torque (N-m), Momentum (N-m-sec)
0.012                         ! Wheel Rotor Inertia, kg-m^2
0.48                          ! Static Imbalance, g-cm
13.7                          ! Dynamic Imbalance, g-cm^2
[... 161 lines cut ...]
0.0   0.0  0.0                !  Initial Xdot, Ydot, Zdot (Non-dimensional)
TRV  "ORB_ID"                 !  TLE, TRV or SPLINE format, Label to find in file
"TRV.txt"                     !  File name
******************* Formation Frame Parameters ************************
L                             !  Formation Frame Fixed in [NL]
0.0  180.0  180.0  123            !  Euler Angles (deg) and Sequence
L                             !  Formation Origin expressed in [NL]
0.0  0.0  0.0                 !  Formation Origin wrt Ref Orbit (m)

using the following command script

<<<<<<<<<<<<<<<<<  42:  Command Script File  >>>>>>>>>>>>>>>>>
0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]

100.0 Align SC[0].B[0] Primary Vector [1.0 0.0 0.0] with L-frame Vector [0.0 1.0 0.0]
100.0 Align SC[0].B[0] Secondary Vector [0.0 0.0 1.0] with L-frame Vector [1.0 0.0 0.0]

# 10.0 SC[0].Thr[2].PulseWidthCmd = 20.0

# 100.0 Point SC[0].B[0] Primary Vector [1.0 0.0 0.0] at World[3]

EOF

the issue still is present and my spacecraft spins throughout the simulation. If I am commenting out align commands and making pointing command active, the spacecraft still rotates randomly not pointing anywhere particularly.
Thank you for responding to my questions. I do appreciate it

ericstoneking, 2022-07-28:
No problem.  I see a couple of things.

1)  Replace KIN_JOINT with DYN_JOINT.  The former is deprecated, and may not be hooked up to anything.  (Sorry about that.)  There is no downside to using DYN_JOINT, even if you don't have any joints.
2)  You have zero wheels and MTBs, but you have inputs for three of each.  This throws 42 out of sync.  If we have zero of something, we leave one in place to serve as a template (see, eg, the Gyro section).  Three is too many.

annamaskolenko, 2022-07-29:
Thanks. I've made changes but no improvements. SC_Simple from Demo folder also does not execute written commands.
The thing I have noticed, however, is that approximately 4000s into simulation both spacecrafts start to rotate in the opposite direction

ericstoneking, 2022-07-29:
I think we're not going to solve this in a chat.  I would recommend that you start over with a clean slate.  Return all input files to their original state.  (Maybe even take the opportunity to download the latest version...)  Confirm that you can run the as-shipped InOut and Demo cases.  Then change one thing and *only* one thing, and confirm that works.  Build back up to your desired configuration one step at a time.

annamaskolenko, 2022-07-29:
Thank you for all your help

closed 2026-02-25 by ericstoneking