ericstoneking/42 #52: Earth twitches on demo

vladtcvs, 2020-11-09:
I have downloaded current master, and have problem - image of Earth twitches. Problem has appeared in commit f84857f016429320324bc00f8e9ab3c34a03fdae

https://youtu.be/-An4PWLio-c

ThatcherC, 2020-11-10:
I get this too! One thing I've also noticed in other simulations affected by this problem is that the spacecraft rotates smoothly the whole time with no twitches, but the background displays the jumpiness seen in @vladtcvs's video.

vladtcvs, 2020-12-05:
It looks I understand why it happens. Celestial bodies, which moves faster, twitche more. I suspect, that when drawing planets camera moves to new point and stays there in sun-centered coordinate system. And planet moves. After some time, camera is moved to follow the planet, but also is fixed in sun coordinate system, and so on