ericstoneking/42 #53: SimReadFromSocket sent and actual

PeteBlanchard, 2020-11-12:
Mr. Stoneking,

Tried to use the socket receiver in 42 to accept a set quaternion value:

```
SC[0].B[0].qn = 0.755903 -0.195576 0.567993 0.260278
```

but it didn't seem to fully use it...the spacecraft did change orientation a bit, but the next reported QN was:

```
SC[0].B[0].qn = 0.725113, -0.196976, 0.576276, 0.321430
```

The value before I issued the command over IPS was:

```
SC[0].B[0].qn = -0.589527, 0.310044, -0.503257, 0.550512
```
Am I using it correctly?  Is there a max rotation it will perform in a step?

PeteBlanchard, 2020-11-12:
Also, is the PosW in ECEF?  Or is it in a different coordinate system?

ericstoneking, 2020-11-13:
Hi Pete,

I just ran the Tx/Rx example, and SC[0].B[0].qn matches up fine.

> Also, is the PosW in ECEF? Or is it in a different coordinate system?

I'm going to leave this one as an exercise for the student :-)

Regards,
-Eric

PeteBlanchard, 2020-11-17:
Thanks, Eric.  I was a little confused looking throughout the codebase...

I found this, which seems to indicate it IS supposed to be ECEF (based on function name):  https://github.com/ericstoneking/42/blob/master/Kit/Source/nrlmsise00kit.c#L2419

From here, looks to be taking the Earth’s CWN and using it to convert from PosN (ECI) to PosW: https://github.com/ericstoneking/42/blob/master/Source/42sensors.c#L387

And this, stating that the CWN is the direction cosine matrix converting from the world-centric inertial frame to the world-centric rotating frame: https://github.com/ericstoneking/42/blob/master/Include/42types.h#L685

But this seems to set it as a rotation about the Z-axis (0 0 1) using the Prime Meridian angle (and not an ECI to ECEF matrix based on time, etc.): https://github.com/ericstoneking/42/blob/master/Source/42init.c#L2620

Regards,

  Pete

ericstoneking, 2020-11-18:
Right.  Where does PriMerAng come from?

PeteBlanchard, 2020-11-18:
LOL got me there, duh (bad student).  Updated comment to ask about nutation and precession...

PeteBlanchard, 2020-12-03:
Eric,

One more question...trying to do some simple ACS type things (using PROTOTYPE_FSW)...was planning on setting the quaternion to do 'point at' type commands, but I see that it does a 'state refresh' each time for those commands (and the sat re-animates each time)... so, I would assume that is NOT the best way to control the satellite...but I didn't want to write reaction wheel controls at this time (and since I had seen the InpCmd 'Point' command, was wondering if there was something like that for the sim socket reader)

Looking at the [SimReadFromSocket](https://github.com/ericstoneking/42/blob/master/Source/IPC/SimReadFromSocket.c#L186), the 'AC.Cmd.Angle' can be set, but I don't see where that is used (off the commit hash I am on; NOS3' "b70508fde8d3973c1b2fccca96bf1ab7294a6588")...

Do I need to write my own 'AcApp' at this point?  Or is there an intermediate approach I could take to do some ACSish controls?

Regards,

  Pete

closed 2022-02-08 by ericstoneking