ericstoneking/42 #57: 42dynamics.c vs 42dynam.c

ThatcherC, 2021-03-31:
Do the files `Source/42dynamics.c` and `Source/42dynam.c` both need to be included in the source code here? I've been working on adding a new component to the dynamics model and these two files have caused me a bit of confusion. It appears that these files are mostly identical, except that `42dynam.c` has some extra code that refers to CMGs. Most of that extra CMG code is excluded with `#if 0 ... #endif` macros. 

I guess the questions that arose while I was poking around in the dynamics are:
- Which of these files should I modify if I want to add a feature of the dynamics model? I've been modifying `42dynamics.c` and that seems to work, but maybe I need to edit both files?
- Do both of these files need to be in the source? Choosing just one of them would make the codebase easier to understand.

Any clarifications here would be greatly appreciated - thanks!

ericstoneking, 2021-03-31:
Hi Thatcher,

Consider 42dynam.c deprecated, replaced by 42dynamics.c.  (Remember the days when filenames had to be eight characters or less?  Old habits take a while to fade.)  

For the record, the CMG code has been subsumed into the multibody code by allowing "embedded momentum" in each body.  There's an example in the Demo.

Regards,
-Eric (he/him)

ThatcherC, 2021-03-31:
Thanks Eric! I'll keep my development restricted to 42dynamics.c then. Is it worth keeping 42dynam.c around? If it's truly just deprecated code, I'd say taking it out would be good practice and a nice way to avoid confusion by other users in the future, but there might be some valuable snippets in there worth keeping that I'm not aware of. What do you think?
Best,
Thatcher

ThatcherC, 2021-03-31:
Awesome. Thanks for clearing this up!

closed 2021-03-31 by ThatcherC