ericstoneking/42 #59: Coarse Sun Sensor `body` initializer

ThatcherC, 2021-05-20:
Hi Eric,

I've been using some coarse sun sensors lately and I'm a bit confused by how the sun sensor axis is set up:

https://github.com/ericstoneking/42/blob/bb9d25d5767ea2df3e4c74736252e70d8c845b77/Demo/SC_ISS.txt#L477-L481

The "axis in body frame" line has 4 numbers, instead of the 3 I'd expect for a unit vector. Looking at `42init.c` shows that the first number of that line is `CSS->Body` and the next 3 are the actual components of the vector:

https://github.com/ericstoneking/42/blob/bb9d25d5767ea2df3e4c74736252e70d8c845b77/Source/42init.c#L2000-L2003

A couple questions on this:
1. What is the `CSS->Body` number? Does it only apply to coarse sun sensors? I don't see a similar `Body` parameter in the `42init.c` lines for the the fine sun sensors, for example:
  https://github.com/ericstoneking/42/blob/bb9d25d5767ea2df3e4c74736252e70d8c845b77/Demo/SC_ISS.txt#L486-L494
2. Does the `Body` parameter need to be defined on the same line as the CSS point vector? I think it would be more clear change the spacecraft definition files to have lines like this:
  ```
  ============================== CSS 0 ==================================== 
  0.1                           ! Sample Time,sec 
  0                             ! Body Index (?)
  1.0  1.0  1.0                 ! Axis expressed in Body Frame
  ```
  That way it's clear which of the numbers the components of the Axis vector.

I'd be happy to put together a PR for item 2 above if you like that suggestion!

Thanks,
Thatcher

ericstoneking, 2021-05-21:
Hi Thatcher,

The Body number is the body that the CSS is mounted on.  I've only had occasion to place CSSs or thrusters on appendage bodies.  The comment on the CSS Axis line should at least be "! Body, Axis expressed in Body frame".  I will take your suggestion and put the Body Index on its own line (for CSSs and thrusters).  Don't bother with a PR.

Regards,
-Eric (he/him)

ThatcherC, 2021-05-25:
Sounds good! Thanks for clearing that up. One other thing - does `Body` as referenced in the CSS definition refer to the same indexing as the `Flex Node Index` in other SC component definitions? I always have Flex Node Index set to 0 for wheels and torquers, like this:

```
*************************** Wheel Parameters ***************************
4                             ! Number of wheels
=============================  Wheel X - CubeWheel Large ================================
0.0                           ! Initial Momentum, N-m-sec
1.0   0.0   0.0               ! Wheel Axis Components, [X, Y, Z]
0.0023   0.03                   ! Max Torque (N-m), Momentum (N-m-sec)
0.0000477                         ! Wheel Rotor Inertia, kg-m^2
0.0                          ! Static Imbalance, g-cm
0.0                          ! Dynamic Imbalance, g-cm^2
0                             ! Flex Node Index
```

```
**************************** MTB Parameters ****************************
3                             ! Number of MTBs
==============================  MTB 0  =================================
1.27                           ! Saturation (A-m^2)
1.0   0.0   0.0               ! MTB Axis Components, [X, Y, Z]
0                             ! Flex Node Index
```

which I always took to mean that these components are mounted on/in Body 0. Are Bodies and Flex Nodes equivalent? It could be advantageous to use the same names for them in the example files if so. And if they're not referring to the same thing, I'd love to learn the difference between them! I haven't found a distinction yet in `Docs/42 Joints.pdf` or `Docs/42 Overview.pdf` but I'm doing some more reading this morning on it.

Thanks!
Thatcher

ericstoneking, 2021-05-26:
Hi Thatcher,

Flex Nodes are a whole different thing.  See Docs/Kane_NBody_Flex.pdf slide 9.  The "few points" mentioned there are flex nodes.  The flex node indices in the SC file refer to the "Analysis Node Locations" defined in the (eg.) InOut/Flex_Simple.txt.

Regards,
-Eric (he/him)

ericstoneking, 2021-08-15:
See today's commit.

closed 2021-08-15 by ericstoneking