ericstoneking/42 #60: "Initial Attitude" for B Axes with respect to L Axes

PeteBlanchard, 2021-05-26:
Mr. Stoneking,

What controls the initial orientation of the spacecraft B Axes with respect to the L Axes?  I see the "Initial Attitude" (in SC_*.txt) angles set to zero's, the rotation quaternion is identity...shouldn't the initial orientation of the spacecraft body axes line up with the local vertical, local horizontal axes?  If not (and it appears not), what controls that?  When I turn the Axes on in the UI, they do not line up.

Looking at a small sat, so only one body, no joints.

Thanks in advance.

ericstoneking, 2021-05-26:
Hi Pete,

Line 13 is the key.  Change that "NAN" to "LAL".

To explain what's going on, line 13 is a three-letter key to how the following three lines should be interpreted.  The first character controls which frame the angular velocity in line 14 is with respect to.  (If it's L, then orbit rate gets added in appropriately.)  The second character (either Q or A) controls whether the [Q]uaternion on line 15 or the Euler [A]ngles on line 16 are used for the initial attitude.  The third character controls which frame the initial attitude is defined with respect to.

Regards,
-Eric (he/him)

PeteBlanchard, 2021-05-26:
ah!  Makes sense, thank you.  As soon as I wrote this I noticed it was aligned with the N Axes but missed the significance of line 13.

Thanks again!

closed 2021-05-26 by PeteBlanchard

PeteBlanchard, 2021-05-26:
Okay, dumb follow up...after I change to LAL, the Y axes (2) line up fine, but the X and Z of the B appear ~180 degrees off (or so) from the L Axes (look to be rotated around the Y).  I don't want to give it an angular momentum and have it spinning...so, I need to compute the quaternion to give it my desired orientation (and change to LQL), correct?

ericstoneking, 2021-05-26:
Hm.  Can you show me that section of your SC file, and a screenshot showing your L and B axes?  (Tip: Pressing 'F1' will save a Cam window screenshot as Screenshots/CamSnap.ppm.)

PeteBlanchard, 2021-05-26:
Here ya go...
[image]
[image]

ericstoneking, 2021-05-26:
Okay, your initial attitude section is good.  Since you're using PROTOTYPE_FSW, now I'm curious about your Inp_Cmd.txt.
Another tip: In Inp_Graphics.txt, line 8, you can make 42 pause at startup, so you can examine the initial condition.  My question is, was the initial attitude wrong, or did the attitude start properly and get steered off by PROTOTYPE_FSW?

PeteBlanchard, 2021-05-26:
Interesting...I switched back to PASSIVE_FSW as a test, and the attitude is still off...

ericstoneking, 2021-05-26:
Sorry, I'm stumped.  It behaves as expected on my setup.  Looks like you've voided the warranty... :-)  Good luck with your bug hunt.

PeteBlanchard, 2021-05-26:
Ha!  No worries, I appreciate the quick responses.  You've been very helpful once again!  Thanks for the information.  :-)