ericstoneking/42 #63: How to get ECEF data from Socket IPC

jwoolston, 2021-08-05:
I have been poking around in the the repository, internet and docs trying to figure out how to obtain ECEF data from the simulation of the socket. Currently what I have achieved is an output packet that looks like:

```
TIME 2014-172-00:00:07.400000000
SC[0].PosR = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].VelR = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].svb = 1.911549569229e-01 7.909220372261e-01 -5.812934830823e-01
SC[0].bvb = -3.996145331303e-05 2.917178560806e-06 -2.245080882139e-05
SC[0].Hvb = 3.102972229380e-02 -1.803329063292e+00 -1.674012035485e-01
SC[0].AC.ParmLoadEnabled = 0
SC[0].AC.ParmDumpEnabled = 0
SC[0].B[0].wn = 3.102972229380e-04 -9.016645316460e-03 -5.580040118285e-04
SC[0].B[0].qn = 5.005973816419e-01 2.828773014671e-01 5.711165756938e-02 8.161623321237e-01
Orb[0].PosN = 5.597997251275e+06 -4.649154587081e+05 3.266666278426e+06
Orb[0].VelN = -3.938776316545e+03 -9.662977797767e+02 6.789562087021e+03
[EOF]
```

From an Inp_IPC.txt that looks like:

```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
WRITEFILE                               ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
SERVER                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
FALSE                                   ! Allow Blocking (i.e. wait on RX)
TRUE                                    ! Echo to stdout
3                                       ! Number of TX prefixes
"SC"                                 ! Prefix 0
"Orb"				! Prefix 1
```

As I understand it, `Orb[0].PosN` is giving me an ECI position and what I want is `Orb[0].PosR` but I am failing to understand why it is not dumping these values. Do I need to adjust `42.json` and re-run the IPC python script followed by rebuild to get this to output or is there something I can change in the configuration? This is all being run from modifications on the Inp_IPC.txt in the Tx folder (i.e. the spacecraft and orbit definitions are whatever is currently there).

ericstoneking, 2021-08-06:
In 42-ese, you're looking for Orb[0].PosW, which doesn't exist but is easy to add.  

1)  In orbkit.h, add a PosW to the OrbitType.  Make sure to add a trailing comment with these characters: [~=~] (see PosN line for example).  These characters tell the python scripts that you want this variable to be available for two-way socket communication.

2)  In 42ephem.c:OrbitMotion, add a line like:
      MxV(World[O->World].CWN,O->PosN,O->PosW);

3)  make clean, make.  The python scripts should add PosW to the socket traffic.

You've probably picked up that "W" is the generalization of the ECEF frame.  Every world has one.

Regards,
-Eric

jwoolston, 2021-08-06:
@ericstoneking Thank you for you speedy response. After making the changes you suggested (and running the python scripts) I was able to get everything working. 

One final question if I may:

Am I correct in my understanding that the `SC[0].B[0].wn` and `.qn` fields are ECI orientation and angular rates of the spacecraft (more specifically Body 0) as well? If so I imagine a similar process would allow me to output a converted version of those in ECEF?

ericstoneking, 2021-08-06:
You have the general idea.  SC[0].B[0].qn is the quaternion expressing the rotation from N to SC[0].B[0].  (SC[0].B[0].CN is the direction cosine matrix for the same rotation).  So you can get qbw by QxQT(SC[0].B[0].qn,qwn,qbw), where you made qwn by C2Q(World[foo].CWN,qwn).

SC[0].B[0].wn is a little different.  It is, as you surmise, the angular rates of SC[0].B[0] with respect to N.  It is expressed, though, in SC[0].B[0], not in N.  In words, I say things like "the angular rate of B with respect to N, expressed in B".  By the time it gets to C, I could have written B.wnb for that, but I dropped the trailing "b" out of fatigue.  Some things are most naturally expressed in B or N, so the "expressed in" part is left implied.  You did the right thing looking in the header; I sometimes leave clues in the comments.  (Based on this conversation, I've tweaked the comment on B.wn to "Angular Velocity of B wrt N expressed in B frame".)

For your B.ww, it's up to you which frame is more natural to express that in.  See mathkit.c and dcmkit.c for handy functions like MxV, QxV, etc.

Regards,
-Eric

jwoolston, 2021-08-06:
@ericstoneking Sounds good. Once again, thank you for your timely response. You have given me enough understanding for my original question and the information for some homework on the rest. I appreciate your time. 

All the best,
Jared

closed 2021-08-06 by jwoolston