ericstoneking/42 #65: Segmentation fault

NEONGASHMEN, 2021-08-23:
Hello,

I'm running 42 on a debian virtual machine. There seems to be no issue in running 42 without graphics, but enabling the GUIFLAG throws a segmentation fault. I'm grad student and a novice in programing. I'm currently working on a CubeSat project and hence came across this. The following are the contents of runlog.txt and makelog.txt  (I've ran <code>make $1 2>&1 | tee makelog.txt</code> and <code>./42 $1 2>&1 | tee runlog.txt</code>, referring to a previous issue)
[makelog.txt](https://github.com/ericstoneking/42/files/7029659/makelog.txt)
[runlog.txt](https://github.com/ericstoneking/42/files/7029660/runlog.txt)

Thank you for your time.

ericstoneking, 2021-08-23:
Hi Neon,

Welcome to the world of debugging!  I highly recommend that you learn the basics of gdb (gnu debugger).  It should help you pinpoint the line where the segfault occurs, which will help you get to root cause.  I don't see anything obviously wrong in your logs.

Regards,
-Eric

NEONGASHMEN, 2021-08-23:
Hi Eric,

Thank you for your support. I'll try to look into it. Can you leave this thread open, so that I can get back if I'm able to find the source of the segmentation fault.

NEONGASHMEN, 2021-08-25:
[image]

Hey Eric,

This is a screen grab from gdb. The segfault seems to originate from glut lib ? Any idea how to mitigate the issue ?

ericstoneking, 2021-08-25:
Sorry, this is outside my scope.  It looks like your operating system and glut aren't getting along.  It could be that updating your glut libs would help, or maybe glfw will work better for you.

NEONGASHMEN, 2021-08-25:
Alright. Thank you Eric, for your help and guidance nevertheless. The issue might be because I'm using a virtual machine. I'll try updating glut libs and try running it on any friends' device.

Thank you for your time

m4v, 2021-09-08:
I reverted to Mesa 21.0.3 (Ubuntu's default) and I have no crashes anymore when using GLUT, GLFW sometimes will crash the same way at startup, in particular with gdb.

For the record, the Mesa libraries I was using was 21.2.1 from kisak-mesa PPA.

closed 2022-02-08 by ericstoneking