ericstoneking/42 #66: Invalid arch name '-arch macOS-arm64'

MilesATurner, 2021-09-06:
Hi Eric, 

I've been enjoying 42 for a bit now, and its helped immensely with running some quick simulations for potential university satellite proposals. I've been using it on a msys, but I  just recently transitioned to a mac with an M1 processor and noticed that the make file keeps failing when trying to run "make" while in 42. After digging through the make file I can't figure out where my issue is or how I can work around it. I have gone ahead and attached a makelog.txt to help show you my issue. Do you have any suggestions? 

Thank,

Miles
[makelog.txt](https://github.com/ericstoneking/42/files/7117932/makelog.txt)

ericstoneking, 2021-09-07:
Hi Miles,

Glad you like it.

Try changing "ARCHFLAG = -arch x86_64" to "ARCHFLAG = -arch arm64", or something like that.  (I don't know your hardware architecture, but from all the warnings, it isn't x86.)  In my Makefile, that's around line 87.

Regards,
-Eric

MilesATurner, 2021-09-07:
Hey Eric,

Looks like that completely fixed the bug!

Thank you again and keep up the good work!

Miles

closed 2021-09-07 by MilesATurner