ericstoneking/42 #68: Error while installing 42 on linux

Korabel47, 2021-10-30:
Hello to all,

I've downloaded a zip-archive from this repository.
While trying to compile the software on Linux Mint 20.04 via command _.../42-master/_**make** I am getting an error about the line _AlbedoWindow = glutCreateWindow("Albedo Scratchpad");_ : 

```

./Source/42glut.c: In function ‘HandoffToGui’:
./Source/42glut.c:1392:7: error: ‘AlbedoWindow’ undeclared (first use in this function)
 1392 |       AlbedoWindow = glutCreateWindow("Albedo Scratchpad");
      |       ^~~~~~~~~~~~
./Source/42glut.c:1392:7: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:271: Object/42glut.o] Error 1
```
Before that, I have had an error with using the glut library inside the code, but resolved it installing the necessary packages. I cannot understand here why am I getting this error on the _AlbedoWindow_ variable is just being declared?

The settings I use in the **.../42-master/Makefile** is (besides that nothing else was changed):
```

GUIFLAG = -D _ENABLE_GUI_
#GUIFLAG =

# For graphics interface, choose GLUT or GLFW GUI libraries
# GLUT is well known, but GLFW is better for newer Mac's hires displays
#GLUT_OR_GLFW = _USE_GLFW_
GLUT_OR_GLFW = _USE_GLUT_

SHADERFLAG = -D _USE_SHADERS_
#SHADERFLAG =

#CFDFLAG =
CFDFLAG = -D _ENABLE_CFD_SLOSH_

#FFTBFLAG =
FFTBFLAG = -D _ENABLE_FFTB_CODE_

#GSFCFLAG =
GSFCFLAG = -D _USE_GSFC_WATERMARK_

#STANDALONEFLAG =
STANDALONEFLAG = -D _AC_STANDALONE_

GMSECFLAG =
#GMSECFLAG = -D _ENABLE_GMSEC_

```

I would very appreciate if somebody directs me to a wholesome 'userguide'.pdf or a FAQ document with tips to make 42 work on my Linux, as I could not find or might miss some corresponding file in the **.../42-master/Docs/** folder. Cannot use the included file README for this purpose.

Best regards

ericstoneking, 2021-11-01:
Hi Korabel,

I'm the user guide, such as it is :-)  

Looks like I left a loose end.  If you have the file Source/42gpgpu.c, then you should see AlbedoWindow initialized (and used) there.  Delete line 1392 from 42glut.c, and see if that solves the issue.  ( I develop on a Mac.  I don't know why it is more tolerant of errors like this than it should be, but sometimes it lets things through that other platforms stumble on.)

By the way, CFDFLAG and FFTBFLAG are hooks into external software that I'm guessing you don't have.  You lose nothing by leaving them blank.  And you might read up on STANDALONE before you enable that one.  Start with 42/Standalone/Readme.txt.

Regards,
-Eric (he/him)

Korabel47, 2021-11-03:
Hello Eric,

Commented out the line 1392; the compilation proceeded further, but bumped into:
```
/usr/bin/ld: ./Object/42ipc.o: in function `InterProcessComm':
~/42-master/./Source/42ipc.c:205: undefined reference to `SendStatesToSpirent`

/usr/bin/ld: ./Object/42perturb.o: in function `Perturbations':
~/42-master/./Source/42perturb.c:1124: undefined reference to `CfdSlosh'
collect2: error: ld returned 1 exit status
make: *** [Makefile:229: 42] Error 1
```

Line 229 in the Makefile:
```
226|##########################  Rules to link 42  #############################
227|
228|   42 : $(42OBJ) $(GUIOBJ) $(SIMIPCOBJ) $(FFTBOBJ) $(SLOSHOBJ) $(KITOBJ) $(ACOBJ) $(GMSECOBJ)
229|	    $(CC) $(LFLAGS) $(GMSECBIN) -o $(EXENAME) $(42OBJ) $(GUIOBJ) $(FFTBOBJ) $(SLOSHOBJ) $(KITOBJ) $(ACOBJ) $(GMSECOBJ) $(SIMIPCOBJ) $(LIBS) $(GMSECLIB)
```

I'm not sure if I need to use the previous approach here. Could you please assist here, too?
Thanks in advance.

Regards

ericstoneking, 2021-11-03:
> By the way, CFDFLAG and FFTBFLAG are hooks into external software that I'm guessing you don't have. 

Looks like I guessed right.  Set these lines back the way you found them:

#CFDFLAG =
CFDFLAG = -D _ENABLE_CFD_SLOSH_

#FFTBFLAG =
FFTBFLAG = -D _ENABLE_FFTB_CODE_

Regards,
-Eric (he/him)

Korabel47, 2021-11-04:
You mean to disable them setting them as ' = <blank>' options? I did that after I read your first answer - reverted the Makefile to its previous state.

Cluena, 2021-11-12:
Hi all, 

I have downloaded the 42 source code, 'make' the folder in Linux and added some libraries as the compiler suggested. 

However, I am getting stuck with -lglfw (glfw library) at Line 127 in Makefile, since the system returns: 
/usr/bin/ld: cannot find -lglfw

Can anyone give me a way to fix this? I have tried some method on google but still not get it done.

Thanks a lot!

ericstoneking, 2021-11-12:
Hi Cluena,

The quickest way forward for you might be to set GLUT_OR_GLFW = _USE_GLUT_.  Look in the Makefile around line 33.  Your linker can't find where your GLFW library is installed, which might mean you need to add a "-L path-to-glfw-lib" into your LFLAGS.  But that'll be moot if GLUT works for you.  

GLFW is an alternative to GLUT.  I've moved to it on my Mac because the newer hires displays don't get along with GLUT.  But on linux, GLUT might still be the way to go.

Regards,
-Eric

Cluena, 2021-11-15:
> Hi Cluena,
>
> The quickest way forward for you might be to set GLUT_OR_GLFW = _USE_GLUT_. Look in the Makefile around line 33. Your linker can't find where your GLFW library is installed, which might [...]

Cluena, 2021-11-15:
> Hi Cluena,
>
> The quickest way forward for you might be to set GLUT_OR_GLFW = _USE_GLUT_. Look in the Makefile around line 33. Your linker can't find where your GLFW library is installed, which might [...]

Thank you Eric. I did not notice the same thing you wrote in README.txt about OpenGL graphics.
It helps me a lot!

ChristopherASuarez, 2021-12-23:
> 

> Hi Cluena,
>
> The quickest way forward for you might be to set GLUT_OR_GLFW = _USE_GLUT_. Look in the Makefile around line 33. Your linker can't find where your GLFW library is installed, which might [...]
It work, thank you Eric.

ThatcherC, 2022-01-06:
I've been having trouble getting 42 to compile with graphics on Ubuntu too. `$ make GUIFLAG=` compiles, but has no graphics enabled (otherwise fine though). I tried `$ make GLUT_OR_GLFW=_USE_GLUT_` (recommended in this discussion) but that fails because `Include/42gl.h` is included in both `42glut.c` and `42gl.c`. 

Could anyone share the process to compile 42 with graphics on Ubuntu? I see that @Cluena and @ChristopherASuarez got it to work but I haven't been able to replicate their successes.

Cluena, 2022-01-07:
> I've been having trouble getting 42 to compile with graphics on Ubuntu too. `$ make GUIFLAG=` compiles, but has no graphics enabled (otherwise fine though). I tried `$ make GLUT_OR_GLFW=_USE_GLUT_` [...]

Did you mean that you can still run 42 and have *.42 output files, but the graphic windows didn't turn on?

In my case, after setting GLUT_OR_GLFW = _USE_GLUT, I also delete line 1392 from 42glut.c as one above comment and then it works.

obsidian-1, 2022-01-08:
Hi!
I am having trouble building 42 on linux. Initially i got a buch of errors with glfw which i fixed by switching to _USE_GLUT_ . Now , it is compiling but not linking , im getting a bunch of these errors.
`/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:100: multiple definition of `Hex'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:100: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:101: multiple definition of `Pent'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:101: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:102: multiple definition of `CosVis'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:102: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:102: multiple definition of `SinVis'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Kit/Include/glkit.h:102: first defined here

/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:65: multiple definition of `FrameRate'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:65: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:65: multiple definition of `RealTimeDT'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:65: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:66: multiple definition of `TimerHasExpired'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:66: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:67: multiple definition of `TimerDuration'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:67: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:69: multiple definition of `MouseClickX'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:69: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:69: multiple definition of `MouseClickY'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:69: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:70: multiple definition of `PauseFlag'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:70: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:71: multiple definition of `PausedByMouse'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:71: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:72: multiple definition of `MouseDown'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:72: first defined here

/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:172: multiple definition of `SphereWindowHeight'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:172: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:173: multiple definition of `CenterWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:173: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:174: multiple definition of `SphereShowWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:174: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:175: multiple definition of `VectorsWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:175: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:176: multiple definition of `FOVsWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:176: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:177: multiple definition of `GridsWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:177: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:178: multiple definition of `AxesWidget'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:178: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:179: multiple definition of `ShowConstellations'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:179: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:181: multiple definition of `PixelScale'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:181: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:183: multiple definition of `TlmIsStatic'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:183: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:184: multiple definition of `UseEphFromTlm'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:184: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:185: multiple definition of `ShowTdrsVis'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:185: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:186: multiple definition of `ShowWatermark'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:186: first defined here
[... cut ...]
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:199: multiple definition of `ColorTexEnabledLoc'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:199: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:200: multiple definition of `BumpTexEnabledLoc'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:200: first defined here
/usr/bin/ld: ./Object/42glut.o:/home/prabhav/Desktop/42/./Include/42gl.h:201: multiple definition of `ReflectEnabledLoc'; ./Object/42gl.o:/home/prabhav/Desktop/42/./Include/42gl.h:201: first defined here
`
Any Help would be appreciated.
@ThatcherC  were you able to build it on ubuntu ?

ThatcherC, 2022-01-09:
@obsidian-1 That's the same error I had! No luck resolving it yet. @Cluena - I wonder if you are perhaps using a slightly out of date? Could you share the commit hash you're able to successfully compile and then I'll try to find which commit between that and the current `HEAD` (which is f20d8d5 at the time of writing) breaks the compile process. Hopefully we can find a good fix soon!

ericstoneking, 2022-02-08:
I cribbed acxz's addition of a glut dependency to the workflow.  I also altered the makefile so unix defaults to glut instead of glfw.  I hope that makes things easier on folks on their first build.

acxz, 2022-02-18:
With the current master (27f4db5) at the time posting, it is important to note that now the newly made default glut backend does not compile on linux systems. And the above PR is required to compile 42.

@ericstoneking I implore you to open this issue (at least to keep a linux support thread open) and to look at #77 at my take on the solution again.

ThatcherC, 2022-02-18:
Cool! All the `make` targets work on my computer now. Haven't run any sims with graphics yet but so far so good.

acxz, 2022-02-18:
Thx so much Mr. Stoneking! While not all `make` targets work for me, `make 42` with either GUI backend works for me know as well! I believe this issue can be closed with the recent commit you have made!

closed 2022-02-22 by ericstoneking