ericstoneking/42 #70: Consideration of GitHub Actions?

ThatcherC, 2021-11-19:
Would the 42 project be open to including some GitHub actions in the repo? I have set up a few in my fork of 42 that run `make` commands every time I push to master (or open a PR) and it's been super valuable. 

As an example, the actions I currently have set up on my fork show that current 42 compiles correctly on Ubuntu, but only with the graphics disabled. GLFW doesn't work on stock ubuntu (maybe there's library I should install) and there's a bug that pops up when you try to compile with GLUT on Ubuntu (probably an easy fix). To me these results show the value of adding Actions - right off the bat I would be able to see if changes I added to 42 would cause build errors on other platforms. I'm going to add Windows and MacOS builds too to see how `make` works on those platforms as well.

**Pros:**
- can test compilation on all platforms when new commits are pushed\
  - will show compilation errors that arise on any of the platforms
- provides reference compilation instructions for all platforms
- requires no C code modifications, just a GH actions description file
- provides maintainers with a quick report on any suggested code changes

**Cons:**
- adds code to repo
- Limited GitHub Action time budget: free accounts are limited to 2000 minutes per month, so if there's a month with lots of pushes to the `master` branch tests might become unavailable toward the end of the month. I don't expect 42 would run into this issue as builds are pretty quick. See [here](https://docs.github.com/en/billing/managing-billing-for-github-actions/about-billing-for-github-actions) for more details

**Example:**

Here's a PR I have open on my fork of 42 (same as upstream 42 but with actions enabled) right now: https://github.com/ThatcherC/42/pull/1. When I submitted this PR, the `make`s automatically ran and identified build errors in the Ubuntu GLFW and GLUT builds and a successful build of 42 without graphics:

[image]

Is this something we could add to the main 42 repo? I could put together a set of Actions if there's interest! I think it'd be a great asset to add the project. 

Best,
Thatcher

ericstoneking, 2021-12-09:
Hi Thatcher,

Yes, I agree that this would be a valuable thing.  It may lead to me actually using github the way it's intended to be used, but maybe that's overdue anyhow.  Please show me what you got.

Thanks,
-Eric

ThatcherC, 2021-12-09:
Awesome! I've opened PR https://github.com/ericstoneking/42/pull/73 that has the new GitHub Actions definition file in it. 

On my fork (https://github.com/ThatcherC/42/), if you click on the "Actions" tab above the usual source code listing, it'll take you to [this page](https://github.com/ThatcherC/42/actions) which shows the results of a few Action runs (including one from today). Clicking on the latest run will show the results - two of the three jobs failed (Linux build with graphics) and one succeeded (Linux without graphics). I'm also able to expand the results of the step of the Action and the results, like the compiler errors that pop up when trying to compile with GLUT. (I'm not sure if you'll be able to see the full the results on my project, but you will be able to if it runs on mainline 42). 

The easiest way to test this would probably to merge the PR (if it looks okay!) and then poke around in the Action outputs.

Best,
Thatcher

closed 2022-02-08 by ericstoneking