ericstoneking/42 #74: LVLH Output

jeffdayhoff, 2021-12-21:
Hello, 

    I am a contractor for the US Space Force. We are using your application for attitude propagation. Would it be possible to easily modify your code to output LVLH? Also, does 42 recommend corrective actions if attitude tolerances are given? 

Regards,
Jeff Dayhoff

ericstoneking, 2021-12-21:
Hi Jeff,

LVLH is easy.  Look in 42report.c to see how to open and write to output files.  If you're referring to the attitude of the spacecraft with respect to LVLH, you may be interested in these lines:

`            
            MxMT(SC[0].B[0].CN,SC[0].CLN,CBL);
            C2A(123,CBL,&Roll,&Pitch,&Yaw);
            fprintf(RPYfile,"%lf %lf %lf\n",Roll*R2D,Pitch*R2D,Yaw*R2D);
`
B[0] is the main body frame, L is LVLH, and N is inertial (Newtonian).  The output file RPY.txt is the attitude of B[0] wrt L expressed as Euler angles (a Body-3 1-2-3 rotation), in degrees.

There's nothing built-in to monitor attitude tolerances or respond to them.  You'll need to roll your own.  I'd suggest 42fsw.c:AdHocFSW as a jumping-off point.

Regards,
-Eric (he/him)

jeffdayhoff, 2021-12-21:
Thank you Eric! That is a big help.

closed 2022-02-08 by ericstoneking