ericstoneking/42 #83: How to compile and Run 42?

nchanilk, 2022-04-14:
Hi 
I'm exploring 42 with Cosmos integration, I'm planning to use 42 on Ubuntu, and I've cloned the project into my Ubuntu server.
How do I compile and Run? Once running, how do I communicate from my client to 42 for input and output? What are the ports opened and what kind of Input and Out put data formats?
Any help would be really appreciated as I'm very new to this field.
Thanks in advance.
Anil

ericstoneking, 2022-04-14:
First off, I know very little about Cosmos or how 42 has been integrated into it, so you may get better guidance elsewhere for that part.

As a standalone, 42 is a C project.  cd to your 42 directory, and enter "make" at the command prompt.  That should compile and link, giving you an executable called 42.  To run it, just enter "42" at the command prompt.

42 runs locally on your machine.  The default folder for inputs and outputs is InOut.  Input files are plain text.  So are output files.  Output files will have a ".42" extension.

From there, read the README.txt, and you're on your way.

nchanilk, 2022-04-14:
Thanks, @ericstoneking .
Sure, I understood the cosmos part.
And your guidance is very helpful and now I'm able to run the make command, and 42. 
Now I'm having an issue with the unable to open display for my system, I'll debug on this as my system issues look like.
BTW, In case of any documentation on Getting started/how-to docs, could you please point me? I'm very new to this tech and trying to understand very basics, and it will be very helpful on any quick start info.

Thanks,
Anil

dmccomas, 2022-05-24:
Hi Anil,

I'm a little behind on email. I'm semi-retired and have some lapses in my
support.

Are you working with OpenSatKit (
https://github.com/OpenSatKit/OpenSatKit/wiki)?  It has an example of
integrating 42 with the core flight system and cosmos. Unfortunately this
is the least documented feature. The Mission FSW QUick StartGuide has a
little information but it's out of date. The architectural components are

1. 42 is compiled so it doesn't use it's 42fsw component and uses a socket
to communicate with an external system
2. OSK's cFS  contains an app called I42 (Interface 42) that reads/writes
data from/to 42 socket
3. I42 communicates to the F42 app (42 FSW) via a shared library. It
doesn't uses the cFS' software bus.
4. F42 implements one of the controllers from the 42fsw.c module. My goal
was to try to use the controller with little to no changes so the shared
library contains 42's AC structure.

Hope this helps. It's on my June list to make sweeping OSK document updates.

- Dave

- Dave