ericstoneking/42 #85: Possible error in accelerometer initialization.

danielgomezberdugo, 2022-05-03:
I believe I found an error in line 2234 of 42init.c. 
Accel->SigV /= 60.0; /* from m/s/rt-hr to m/s/rt-sec */
Should be
Accel->SigV /= 3600.0; /* from m/s/rt-hr to m/s/rt-sec */

Let me know if there is a better way to post this kind of find.

ericstoneking, 2022-05-03:
This is as good a place as any to raise issues.

Units on these terms are always a pain to work through, and I have to work at it every time.  So you might be right, but I'd appreciate it if you'd lay out your reasoning.  I do want to make sure that you read the "rt" in "m/s/rt-sec" as "square root".

danielgomezberdugo, 2022-05-03:
I see now, in this case dividing by 60 is correct. It is true that this units are bit of a pain. Thank you for the prompt response.

closed 2022-05-03 by danielgomezberdugo