ericstoneking/42 #87: Companies using 42 Simulator

satzguyz, 2022-05-04:
Hello,

Just curious, what is the flight heritage of the 42 ADCS simulator. Are there any companies already using it in their development cycle?

ericstoneking, 2022-05-04:
Companies?  I don't know.  If anyone would like to share who they are and what they're using 42 for, I reckon I can try starting up a wiki page.

42 is my personal workhorse simulation tool.  I've used it in some capacity on the Fermi Gamma ray Space Telescope, OSIRIS-REx, Constellation, and many proposal studies.  It is currently being used (with custom modifications, of course) as the ACS high-fidelity simulation tool for the Roman Space Telescope, which is currently in Phase C-D.

It is also being used by some cubesat projects here at NASA GSFC, and I know that it's been incorporated into NOS3 and OpenSatKit.  Judging from some questions I get, I know that some university labs have at least tried it out.

satzguyz, 2022-05-05:
Thanks eric. That sounds really interesting. Also, NOS3 and OpenSatKit both
looks similar as they are using 42, cFS and Cosmos. Any differences you
come across Or any recommendations?

djumpa, 2022-05-27:
I have started to use 42 recently and wondered about flight heritage too. It has become my main go to simulator when it comes to fsw. It would be helpful to have information on wiki, as eric suggested. 

I am curious, what kind of custom modifications you made to qualify as ACS high-fidelity simulation tool?

Thanks!

TenguBoss, 2024-03-11:
This thread is almost 2yo but I have decided to jump in with the hope of reviving the conversation. I am an engineer at a small company that works on on-board autonomy for rendezvous and proximity operations. We have just submitted a proposal to the 2024 NASA SBIR solicitation where we plan to employ 42 to test relative state estimation algorithms in a software-in-the-loop environment. 

@djumpa, do you mind sharing what your requirements are for a "high-fidelity simulation tool?" Par example, are you looking for a simulator to verify attitude knowledge requirements to miliarcsec? 

Thank you.
BU