ericstoneking/42 #88: Segmentation Fault

satzguyz, 2022-05-21:
Hello,

I am facing segmentation fault while running ADCS simulator.

> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> 0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
> Reached CmdScript EOF at Time = 0.000000
> Initializing GLUT
> Initializing Cam Window
> [New Thread 0x7ffff2480700 (LWP 20586)]
> Loading Cam Shaders
> Loading Cam Textures
> Loading 3D Noise
> Loading Cam Lists
> Cam Window Width = 800
> Cam Window Height = 800
> Cam Screen Width = 1600
> Cam Screen Height = 900
> Done Initializing Cam Window
> 
> Thread 1 "42" received signal SIGSEGV, Segmentation fault.
> 0x00007ffff3197af8 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> (gdb) bt
> #0  0x00007ffff3197af8 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #1  0x00007ffff295de79 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #2  0x00007ffff2998af6 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #3  0x00007ffff298d862 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #4  0x00007ffff28b92c6 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #5  0x00007ffff2c0c3fe in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
> #6  0x00007ffff7d83492 in glutBitmapCharacter () from /lib/x86_64-linux-gnu/libglut.so.3
> #7  0x00005555555e3717 in DrawBitmapString (font=0x7ffff7fb38c0 <glutBitmap8By13>, string=0x7fffffffdac0 "Sim Time =     0.00") at ./Kit/Source/glkit.c:37
> #8  0x00005555555c0e08 in DrawCamHUD () at ./Source/42gl.c:705
> #9  0x00005555555c62dc in DrawNearAuxObjects () at ./Source/42gl.c:1658
> #10 0x00005555555cca8a in CamRenderExec () at ./Source/42gl.c:2521
> #11 0x00007ffff7d8b194 in ?? () from /lib/x86_64-linux-gnu/libglut.so.3
> #12 0x00007ffff7d8ec39 in fgEnumWindows () from /lib/x86_64-linux-gnu/libglut.so.3
> #13 0x00007ffff7d8b7bd in glutMainLoopEvent () from /lib/x86_64-linux-gnu/libglut.so.3
> #14 0x00007ffff7d8bff5 in glutMainLoop () from /lib/x86_64-linux-gnu/libglut.so.3
> #15 0x00005555555e36da in HandoffToGui (argc=1, argv=0x7fffffffe088) at ./Source/42glut.c:1399
> #16 0x000055555555b60c in exec (argc=1, argv=0x7fffffffe088) at ./Source/42exec.c:401
> #17 0x0000555555559ccd in main (argc=1, argv=0x7fffffffe088) at ./Source/42main.c:41
> (gdb) 
>

ericstoneking, 2022-05-23:
Thanks for including the backtrace.  I can't reproduce the problem.  My best guess (which may be obvious) is an incompatibility with your glut library, so I'd recommend starting there.  The string looks okay, so the problem might be the font, which comes from the library.  FWIW, I use freeglut when on linux.  Good hunting.

ericstoneking, 2022-06-06:
This is getting beyond my area of competency (I'm a dynamicist, not a computer guy).  But seeing "dri" and "so" in the complaining filenames makes me think that your graphics drivers and other GUI infrastructure aren't getting along.  Update everything?

It might be helpful to figure out what line in 42 triggered the segfault.  Since backtrace failed, you might try sprinkling printf statements around, or single-stepping to isolate the error.