ericstoneking/42 #91: Reaction Wheels

annamaskolenko, 2022-07-21:
Hello. I am a bit struggling on how to utilise reaction wheels to stabilise the SC. I have defined several in the SC description txt file, however the SC still rotates. I wanted to align some B axes with L axes by writing the 'align' function in the command file but this also did not work. Should actuators be activated via the Inp_Cmd.txt file? If yes what function should be used? Thanks

annamaskolenko, 2022-07-22:
Thank you for your answer. I tried to run Aura, which uses THREE_AXIS_FSW and has wheels and MTBs defined in its description file, just to play with the spacecraft's stabilisation and aligning the SC with L-frame, but Aura rotates throughout the simulation. Am I missing something?

ericstoneking, 2022-07-22:
Hi,

Try this:
1)  Copy SC_Aura.txt and Orb_EOS.txt from the Demo folder to the InOut folder.
2)  In InOut/Inp_Sim.txt, replace "Orb_LEO.txt" with "Orb_EOS.txt" and "SC_Simple.txt" with "SC_Aura.txt".  Save.
3)  Run 42 (with no arguments).

When I do this, it looks fine.  Aura holds LVLH.  The solar array takes about 10 minutes to acquire the Sun, and then tracks it.  If this doesn't work for you, then something is amiss.

Regards,
-Eric

annamaskolenko, 2022-07-22:
Thanks! Will try this. Also, if changing the fsw for the cubesat from prototype to three-axis, 42 does not run at all. I have defined the wheels and MTBs for the cubesat and made it 'look similar' to Aura (in terms of the chosen characteristics), but 42 does not want to run cubesat in a new fsw

closed 2022-09-14 by annamaskolenko