ericstoneking/42 #93: Communication with external Application

YukiImai-Tengun, 2022-07-28:
Hello.
I have a question regarding the communication method between 42 and an external application.
I would like to send values to 42 from an external application for each step, run a simulation reflecting the values for each step, and send back the results for each step.
I have modified the file Inp_IPC.txt to run 42 as a server in "TXRX mode" and send values from an external client to 42 like "SC[0].AC.Whl[0].Tcmd = xx".
In fact, the transmission succeeded, and we confirmed that the value we sent was stored in a variable in SimReadFromSocket.c. However, according to the returned simulation results, the velocity around the axis was hardly changed." No matter what value was used for the "xx", the result remained the same. In other words, it appears that the torque values sent were not reflected in the simulation.
Is the communication setting appropriate, i.e., is it correct to set it to "TXRX mode"?

Thanks,
Yuki

ericstoneking, 2022-07-28:
Hi Yuki,

It sounds to me like you have the basics correct.  I would ask two questions at this point:
1)  Is your external client running your control law?  If this is the case (and I guess it might be if you are sending wheel torque commands back to 42), then you might try the "ACS" option instead of "TXRX".  The "ACS" option is special because the IPC happens in a different part of the sim step (browse 42exec.c and 42fsw.c to see where).  It's possible that the values you provide are being overwritten.
2)  This may sound like a dumb question, but does your S/C have wheels?  If not, the torque command has nowhere to go.

mhmodayman, 2023-05-21:
@YukiImai-Tengun
How did you send external control cmd? do you add [EOF] or '\n" at the end of the line?
can you share with me the syntax you used, please?