ericstoneking/42 #96: Commands within THR_FSW

annamaskolenko, 2022-08-25:
Hello Eric,
Hope I am not bothering you too much.
I have a question regarding the Thruster FSW.
Is there a way to command SC's attitude via a command file?
I have found these lines in 42fsw.c in ThrFSW section from the source files, and I believe these command the attitude of the SC, because when I change these SC behaves as expected.
      double RollCmd[4] = {30.0,0.0,-30.0,0.0};
      double PitchCmd[4] = {0.0,30.0,0.0,-30.0};
      double YawCmd[4] = {0.0,0.0,0.0,0.0};
Also, when I change MoveTime from 1000.0 to another value thrusters get activated as expected as well.
Lastly, how can I output data from thrusters, i.e. thrust force at a particular point of time?
Thank you in advance,
Anna

ericstoneking, 2022-08-28:
Hi Anna,

ThrFSW uses local variables for the purpose of demonstration.  You can modify that code however you like.  If you would like to inject commands from Inp_Cmd.txt, there are a couple of things you need to do:
1)  Put some global variables into the struct AcThrCtrlType definition in AcTypes.h to hold your command values.
2)  Add some code to 42fsw.c:FswCmdInterpreter to read a command from Inp_Cmd.txt and populate your new global variables.
3)  Modify ThrFSW to use your new variables.

Depending on your needs, it might be simpler just to put your command logic within ThrFSW and skip the Inp_Cmd.txt.  Whatever works best for you.

To output thruster data, look in 42report.c to see how global variables are output.  Thruster force is defined in 42types.h in the ThrType struct.  The scalar thruster force itself is just the variable F.

Regards,
-Eric

OEB1, 2022-12-16:
Hi Eric

I've been looking at this code with Anna, and I also have a question or two about the ThrFSW if you don't mind taking a look.

We've tested the code as you've written it, and it works as expected.  However, I have also tried to command just the attitude with no translation.  In this case, when you set just the angles, the thrusters don't fire.

From what I can understand  from  lines 1637 of 42fsw.c

         if ( FoA > 0.0 && TorxA > 0.0) {
            T->PulseWidthCmd = (0.25*FoA + TorxA)/T->Fmax*AC->DT;
         }

the FoA term comes from the translation requirement, and the TorxA term comes from the rotation requirement.

As FoA is always zero if you don't ask for a translation, the thrusters never fire.

I've changed my version of the subroutine to 

 if ( FoA >= 0.0 && TorxA >= 0.0) {...

It seems to work when I test it, but worried I've misunderstood something - maybe it will be less stable when controlling the attitude.  Do you have any thoughts about what I've done?

When we resolve this we can move onto getting the commands from the command file into our version of the FSW.

Thanks!

Oliver

ericstoneking, 2022-12-16:
Hi Oliver,

What you've done is fine, for what you're working with.  The code you're working from is a very simple-minded rule for demonstration purposes.  Distributing force and torque commands to thrusters is a topic all by itself, and there are several ways to tackle the problem.  If you want to get into that, maybe we should talk offline.  If you're just looking to get on with the next step, I'd say go for it.

Regards,
-Eric

OEB1, 2022-12-16:
Thanks for quick reply. For now I think this example code will help us.

As and when we need to develop more complex controls I may be back in touch.

Thanks again!

Oliver