ericstoneking/42 #97: More demos and documentations

PieDema96, 2022-09-01:
Hi there!
I was wondering if anybody knows where to find more examples for 42 tools?  In particular I am looking for something tightly related to GN&C operations.

ericstoneking, 2022-09-12:
Check out OpenSatKit and NOS3.  Both of those use 42 as a component of a larger architecture, and both are geared for ops.

dmccomas, 2022-09-21:
OpenSatKit (OSK) is located at https://github.com/OpenSatKit/OpenSatKit/wiki

OSK is a work in progress and one ultimate goa is to provide a core Flight System reference mission. Some of the pieces ae in place that may hep you.  42's ThreeAxisFSW() from acfsw.c is re-hosted in a cFS app called F42. A second cFS app called I42 manages the cFS to 42 interface.  

SimSat is a fictitious mission that is being defined and implemented in OSK. There are pieces in place like generating science files and downlinking them during a comm pass. SimSat has not been combined with the 42 closed loop example, but the ultimate goal is to make a complete reference mission.