ericstoneking/42 #99: Reporting Ideal Actuators

annamaskolenko, 2022-09-14:
Hello Eric
I was trying to output some data from ideal actuators, which are used by prototype fsw, and got stuck.
Here are the lines I have inputted into the appropriate sections of 42report.c file.

static FILE *IdealTrqfile;

IdealTrqfile = FileOpen(InOutPath,"IdealTrq.42","w");

fprintf(IdealTrqfile,"\n",SC[0].Tcmd);

The error I am getting says that there is no such a member called Tcmd in struct SCType struct. 
I was wondering how I can call Tcmd out from IdealActType struct instead of SCType struct. 
I would be grateful if you could help me. Thank you in advance.
Anna

annamaskolenko, 2022-09-14:
UPDATE

I have changed the third line to      fprintf(IdealTrqfile,"\n",SC[0].IdealAct[3].Tcmd);      but this outputted a blank IdealTrq.42 file

ericstoneking, 2022-09-14:
Hi Anna,

Try this:

fprintf(IdealTrqfile,"%lf  %lf %lf\n",SC[0].IdealAct[0].Tcmd,SC[0].IdealAct[1].Tcmd,SC[0].IdealAct[2].Tcmd);

The Tcmd you want is in struct IdealActType.   Also, read up on fprintf format strings.

Regards,
-Eric (he/him)

annamaskolenko, 2022-09-14:
Thank you very much. That helped.
I am just wondering why exactly three torque components are used. Are these associated with three ideal reaction wheels?
Also, it is interesting that as the SC is stabilised torque values go to 0. I thought torque should have remained to always adjust the SC to meet the commanded attitude. Or is it connected with environmental forces set to zero?

ericstoneking, 2022-09-14:
Three ideal actuators aligned with the three body axes.  If you want more complicated than that, it's time to move up to real actuators :-)

You surmise correctly that non-zero environmental torques would lead to non-zero steady-state torque commands.   As it is, there's nothing to push you off zero error, so the control torques can approach zero.

closed 2022-09-21 by annamaskolenko