ericstoneking/42 #105: 42 in WSL2 on Windows was very smooth

mmisbach, 2023-01-06:
Running 42 on windows using WSL2 was surprisingly easy. All I did was follow Microsoft's directions [here](https://learn.microsoft.com/en-us/windows/wsl/tutorials/gui-apps) which are basically just install the appropriate vGPU driver and install/update WSL2. From there, I was able to open the linux/WSL prompt, apt install the needed dependencies, and compile and run 42 without any need for Xming/Xlaunch and messing around with display stuff like in the "Install Ubuntu on Win10.txt". 

I am on windows 11 but the Microsoft documentation says windows 10 after a certain update should be the same. I'm just getting started but so far everything seems to be working as expected. Maybe WSL2 should be the recommended route for Windows users?

ericstoneking, 2023-01-09:
Thanks for the good news.  I am migrating to recommending WSL over msys.  This will make that easier.

clxyder, 2023-05-20:
Just a heads up, this is with respect to remote development, sshing into a machine with the `wsl --update` feature will not work.

https://github.com/microsoft/WSL/issues/9231