ericstoneking/42 #107: Electrical thruster modelling

PieDema96, 2023-02-09:
Hi there!

I am working on a **42** based scenario where I would like to model an electrical thruster attached to a satellite core.
I have noticed that in 42 it is possible to set up a custom force applied at the core centre of mass, but I am not able to add a real hardware thruster. Is there this possibility in NASA-42? Is it possible to implement similar hardware in an environment like NOS3 or similar, powered by 42?

Thank you in advance!

PieDema96, 2023-02-10:
Sorry if my question was not completely clear. I'll try to clarify. I don't need to have a HW-in-the-loop thruster but I simply want to model an electric thruster inside the 42 simulations.
I know that I should add it inside the SC_.txt file, however, I must model an electric thruster, and I am not sure that 42 is able to do so. In my understanding, 42 only support impulsive thrusters while a need an electric thruster that delivers a continuous force. Am I missing something on 42 modelling capabilities?

Thank you in advance

ericstoneking, 2023-02-10:
Thanks for the clarification, it helps me a lot.  I *think* I understand what you're looking for now.  Let's find out:

I just pushed an update.  You may now designate a thruster as PULSED or PROPORTIONAL.  To command a proportional thruster, populate the variable Ac.Thr.ThrustLevelCmd with a value between 0.0 and 1.0.  Is that the capability you're after?

Regards,
-Eric (he/him)

PieDema96, 2023-02-13:
Thank you so much for the quick update and kind answer! Very appreciated.

However, I am looking for a different model I guess. I am really looking for a continuous thruster action in time, and not punctual/pulsed. I would like to have an electric propulsion acting as a force I can constantly apply to the satellite along a certain time span and eventually update (changing the module and orientation).

Please let me know if I am still not clear enough.

Best regards,
-P

ericstoneking, 2023-02-13:
Hi P,

Okay, it sounds like you can do that with either a pulsed thruster (that's pulsed continuously on) or a proportional thruster (that's set at the force level you want and left alone thereafter).  Is the question then how to command a thruster?  The usual way is to have a flight software model (in 42fsw.c) that sets variables (like SC[0].Ac.Thr[0].ThrustLevelCmd).  You can also use the command script capability.  Browse 42fsw.c:FswCmdInterpreter() and Inp_Cmd.txt for ideas on that.  Either way, you'll need to do some C programming, but it could be worse.

Regards,
-Eric (he/him)

PieDema96, 2023-02-14:
Dear Eric,

Thanks a lot again for the insights.
Got your point. However, observing the models in 42 I get that both pulsed and proportional apply the thrust action instantaneously at the time indicated in the Inp_Cmd.txt file. And it seems to me that this is just an instantaneous effect and is not applied throughout the entire step; let's say I write in the file:
- apply 10N at  0s
- apply 10N (or a different value) at 10s
between 0 and 10 sec is the action continuous or just instantaneous at the beginning of the following interval? About that, I am looking for the capability of applying a constant action along the interval 0-10 s (in this case).
Sorry for being punctilious, just trying to understand better.

Thank you in advance for your time.
Best regards,

P-

ericstoneking, 2023-02-14:
See if the attachment sheds some light.

[ForPiedema.pdf](https://github.com/ericstoneking/42/files/10733686/ForPiedema.pdf)

msuder, 2023-02-15:
> See if the attachment sheds some light.
> 
> [ForPiedema.pdf](https://github.com/ericstoneking/42/files/10733686/ForPiedema.pdf)

Hi Eric... I found the attachment very useful for my understanding... will you consider adding it permanently to the 42/Docs directory in git?

ericstoneking, 2023-02-15:
Hi Mark,

Yes, I'll neaten it up a bit and put it in the Docs folder.

Regards,
-Eric (he/him)

closed 2026-02-25 by ericstoneking