ericstoneking/42 #108: Setting spin rate in Prototype FSW

annamaskolenko, 2023-03-02:
Hi Eric,

I have tried to set a spin rate for my spacecraft and found out that the following input command is not being read

"%lf Spin SC[%ld] about Primary Vector at %lf deg/sec"

Setting the time stamp to 0.0 makes the spacecraft to stall. Setting the timestamp to any other value leads to the command being ignored and skipped over.
Do you have any thoughts on what might be going wrong?

Thank you,
Anna

ericstoneking, 2023-03-03:
Hi Anna,

Have you defined a direction for the Primary Vector?  See if this works:

0.0 Point SC[0].B[0] Primary Vector [0.0 0.0 1.0] at SUN
0.0 Spin SC[0] about Primary Vector at 1.0 deg/sec

Regards,
-Eric

ericstoneking, 2023-03-09:
It's not recommended until you know what you're getting yourself into.  By which time you should have outgrown PROTOTYPE_FSW.