ericstoneking/42 #109: Control step length

fede-paga, 2023-03-03:
Hi!
I'm using 42 to validate a low-thrust orbit controller, but I'm facing some issues. I'd like to execute the validation in the Standalone working mode, by communicating to the 42 simulation engine the thrust control action to be applied at each control step. I managed to set up the communication as shown in the example, and I have a working 42-custom controller closed loop. However, I still have some doubts:

1. Is it possible to change the control step to a different value than the time step? I only see the _Sim Duration_ and _Step Size_ parameters in the input file _Inp_Sim.txt_ but I am not able to choose a control step of the thrust action communicated by my controller different than the Step Size
2. How can I specify that the incoming control action from my controller should be interpreted as a continuous thrust? From my understanding, 42 interprets the thruster force command as an impulse that is applied only in the current time step. However, I'd like to have a continuous control action, repeated in each time step inside the control horizon to model an electric thruster. Is it possible to do so?
3. How can I control the attitude to have a S/C always aligned with the RTN frame during the simulation? Since my control actions are referred to the RTN frame, I need an ideal attitude control to enforce that. But I am not sure how I can enforce the alignment with the RTN frame while commanding the custom thrust action communicated by my Standalone controller

Thank you very much for any suggestions.
Fede

ericstoneking, 2023-03-03:
Hi Fede,

1)  Look in your SC_foo.txt file, line 6 for your FSW sample time.
2)  I'm guessing your installation is more than three weeks old :-)  By request, I recently made a clear distinction between PULSED and PROPORTIONAL thrusters.  See Docs/Thruster Modes.pdf for more.
3)  What does RTN signify?

Regards,
-Eric

fede-paga, 2023-03-03:
Hi Eric,

 thank you very much for your answers. I totally missed those two points...

Regarding the third point, I'd like to align the body axis of the S/C with the radial, along-track, cross-track frame (that is the LVLH frame). However, the torque commands needed to control the attitude are not provided by my external controller. If it's possible, I'd like to use one of the controllers already implemented in 42 on top of my orbit controller, but I'm not sure on how to do that. I'd like to call the function `AcFsw(&S->AC)` together with my controller, is there a standard way to do that? 

Thanks,
 Fede

ericstoneking, 2023-03-03:
Hi Fede,

Radial/Track/Normal.  Makes sense, I'd just never run across the acronym before.  So in 42, the "L" frame is a permutation of it, so that part is easy enough.

There is no "standard" way, but I think you can cobble one together without too much trouble.  Look in 42fsw.c:AdHocFSW().  It happens to be controlling to a permutation of L.  

I'm a little unclear about your external controller, but if you're talking to it over a socket, then the call to it is in 42fsw.c:FlightSoftWare() in the CFS_FSW case.  You can add a call near there to call your attitude controller, either AdHocFSW or something similar.  Or you can add code to your external controller, either way.

Refer back to Docs/42 Overview.pdf and Docs/Nomenclature.pdf if you need to decypher reference frame notation.

Regards,
-Eric

fede-paga, 2023-03-07:
Hi Eric,

 thank you very much for your helpful answer. I still have a doubt though. What if I need to operate the two controllers at different frequencies? It appears to me that I can call the AdHocFSW at each simulation step (instead of only at each FSW step) to enforce a higher frequency for the attitude control, but is it a good practice? I have some trouble in understanding the best approach to decouple attitude and orbit controls.

Thanks
-Fede

ericstoneking, 2023-03-07:
Hi Fede,

Study the code.  If you can understand how the FSW and sensors run at different sample times than the sim timestep, then you can figure out how to make your controller(s) run at the sample rate of your choosing.  

Regards,
-Eric