ericstoneking/42 #111: Detumble process

LeoVLU, 2023-03-17:
Hi Eric!
I’m trying to add detumbling process using MTB in 42 but i’m having some questions.
1. Does 42 support to simulate the process?
2. I’d like send control command from cFS which communicates with 42 through socket, what should i do with 42. Till now, i have no clues at all.
3 Do i need to create new FSW or just need to modify the cFS_FSW?

ericstoneking, 2023-03-31:
Is this a question about how 42 works, or a request for engineering advice?  If the former, please describe your issue in detail.  The latter is out of scope until I retire from my day job.  Look me up in about twelve years :-)

LeoVLU, 2023-04-05:
:)))He He. Apologize for my ambigugous question. I did run successfully the Detumble process relating with MTB Model in 42. But it happend only when i changed time mode from Fast to External. I assume the problem may be caused because i set up the algorithm in external app to caculate the Mcmd and used socket to transfer data between them . Can you verify my hypothesis. Have a nice day.