ericstoneking/42 #113: Rx from python

mhmodayman, 2023-03-26:
I am trying to receive data from Tx using a python script as the Rx (both on the same device)

Here is the code for Rx, I wrote a code that greps over available ports
It can grep over all ports except for 10001, and 44387

```
import socket

def create_socket(port_number):
    server_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    server_socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    server_socket.bind(('127.0.0.1', port_number))
    server_socket.listen(1)

    return server_socket

socket_list = []

for port_number in range(1025,65536):
    try:
        socket_list.append(create_socket(port_number))
    except Exception: 
        print("didn't listen to ", port_number)

```

I am not sure how to receive from Tx, am I doing something wrong?

This is the configuration for Tx
```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
RX                                      ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
CLIENT                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
TRUE                                    ! Allow Blocking (i.e. wait on RX)
TRUE                                    ! Echo to stdout
0                                       ! Number of TX prefixes
```

mhmodayman, 2023-03-27:
Okay will try this fix, but not both ends of the socket are TX. one is TX, and the other are RX, but the thing is that Tx and Rx ends are on the same PC device, would that be an issue?

ericstoneking, 2023-03-27:
No, both ends of the socket can be on the same device with no problem.  You have run the TxRx example problem, right?

mhmodayman, 2023-03-28:
Yes I tried TxRx example, and it works

Here is the new configuration for Inp_IPC, but still the same issue

```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
TX                                      ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
SERVER                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
TRUE                                    ! Allow Blocking (i.e. wait on RX)
TRUE                                    ! Echo to stdout
1                                       ! Number of TX prefixes
"SC"                                    ! Prefix 0
```

mhmodayman, 2023-03-29:
Okay sure.

Meanwhile, could you please confirm the following configuration for the TxRx example?

Tx:
- it is the socket server
- its host is 127.0.0.1
- its port is 10001

Rx:
- it is the socket client
- its host is 127.0.0.1
- its port is 10001

Correct?

mhmodayman, 2023-03-29:
It is working now

I had to replace this line: client_socket.bind((client_host, client_port))
with this: client_socket.connect((client_host, client_port))

```
import socket

client_port = 10001
client_host = "127.0.0.1"
client_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM , 0)  # AF_INET: IPv4, SOCK_DGRAM: UDP, SOCK_STREAM: TCP
client_socket.setblocking(False)
client_socket.settimeout(10)
client_socket.connect((client_host, client_port))

try:
    read_bytes = client_socket.recv(1)
except socket.timeout:
    print('WARNING Timedout on receiving bytes from ethernet')
except OSError:
    print('Server not connected yet')

client_socket.close()
```