ericstoneking/42 #116: Euler angles and angular rates wrt orbit reference frame

mhmodayman, 2023-04-11:
Hello Eric

I am using socket programming to interface with your simulator. Your simulator is the TX side.

I am struggling to find how can I report using socket:
1- euler angles of the main body with respect to orbit reference frame
2- angular rates of the main body with respect to orbit reference frame
3- I want to confirm if orbit reference frame is the same as LVLH in your simulator

This is the Inp_IPC file

```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
TX                                      ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
SERVER                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
TRUE                                    ! Allow Blocking (i.e. wait on RX)
TRUE                                    ! Echo to stdout
3                                       ! Number of TX prefixes
"SC"                                    ! Prefix 0
"Orb"                                   ! Prefix 1
"World"                                 ! Prefix 2
```

and this is the output I am getting as of now
```
TIME 2023-091-00:00:02.000000000
SC[0].PosR = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].VelR = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].svb = 1.597590108428e-01 1.018398953365e-01 -9.818888400284e-01
SC[0].bvb = 0.000000000000e+00 0.000000000000e+00 -0.000000000000e+00
SC[0].Hvb = 7.088104198251e-05 -1.046206690504e-03 3.568393131993e-05
SC[0].AC.ParmLoadEnabled = 0
SC[0].AC.ParmDumpEnabled = 0
SC[0].B[0].wn = 7.875671331390e-05 -1.162451878338e-03 1.189464377331e-04
SC[0].B[0].qn = 2.062720814948e-01 6.768945928731e-01 -2.076632127589e-01 -6.753825054009e-01
SC[0].GN.Pos = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].GN.PosRate = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].GN.Ang = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
SC[0].GN.AngRate = 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00
Orb[0].PosN = 6.978128628534e+06 8.452597279137e+03 1.253149080861e+04
Orb[0].VelN = -1.637145984590e+01 4.226292029335e+03 6.265735604229e+03
World[3].PosH = -1.467010361715e+11 -2.856012611759e+10 0.000000000000e+00
World[3].eph.PosN = -1.467010361715e+11 -2.856012611759e+10 0.000000000000e+00
World[3].eph.VelN = 5.208518237659e+03 -2.935490158302e+04 -0.000000000000e+00
World[10].PosH = -1.469742185582e+11 -2.826368103144e+10 3.650135433286e+07
World[10].eph.PosN = -2.731823867180e+08 2.574636440388e+08 1.514084210472e+08
World[10].eph.VelN = -7.076089727499e+02 -6.008000724794e+02 -2.750860587448e+02
[EOF]
```

mhmodayman, 2023-04-13:
Hi Eric,

I find these variables are already in the 42types.h
SC[0].B[0].CN
SC[0].CLN
SC[0].B[0].wn
SC[0].wln
so I think I don't need to run "python HeadersToJson.py" and "python JsonToIPC.py", right?

I understood (from Nomenclature.pdf) that for Euler angles with respect to L, I can use the following, right?
```
double CBL[3][3], Roll, Pitch, Yaw;

MxMT(SC[0].CLN, SC[0].B[0].CN, CBL);
C2A(321,CBL,&Yaw,&Pitch,&Roll);           % results in euler angles represented in L frame.
```
But I am not sure where should I add these lines (to define them).
and where should I add them to be sent over socket (I assume they should be in SimWriteToSocket.c).

and (from mathkit.c) I understood that for angular rates of B with respect to L, I can use the following, right?
```
double wbn_L[3];                 % angular velocity of B in N (expressed in L)
double wln_L[3];                 % Angular velocity of L in N (expressed in L)
double wbl_L[3];                 % Angular velocity of B in L (expressed in L)
double CNL[3][3];                % DCM of N in L (or from L to N)

MxV(CBL, SC[0].B[0].wn, wbn_L);  % CBL that I calculated above for Euler angles
MT(SC[0].CLN, CNL);
MxV(CNL, SC[0].wln, wln_L);

wbl_L[0] = wln_L[0] - wbn_L[0];
wbl_L[1] = wln_L[1] - wbn_L[1];
wbl_L[2] = wln_L[2] - wbn_L[2];
```
But I am not sure where should I add these lines (to define them).
and where should I add them to be sent over socket (I assume they should be in SimWriteToSocket.c).

ericstoneking, 2023-04-13:
Hi Mahmoud,

There are a couple of ways to go about this.  One, you could add your lines directly into SimWriteToSocket.c.  This keeps all your changes local, which is nice.  The downside is that SimWriteToSocket.c, and all the files in the Source/IPC folder, get rewritten by the python scripts.  If you never run the python scripts, no problem.  If you do, you'll lose any hand-written code you've added.

The other way works within the paradigm that uses the python scripts to autogenerate all the IPC code, so it's preferable if you're going to be adding more things in the future (and for any third party reading this thread later).

First, add your new variables to the SC struct definition in 42types.h.  So, for example, you'll have a SC[0].wbl_L.  Annotate the definition with the [\~=\~] string.  Then run the python scripts, make clean, remake.  Your new variable will then appear in SimWriteToSocket.c along with all the others.

You still need to actually compute your new variable.  You have choices on where to do this; I have no strong suggestions.  I might put a function in 42ipc.c, or in 42report.c.  Just somewhere convenient that's after all your inputs are freshly computed, and before the socket gets written to.  See 42exec.c:SimStep to see the order things get called in.

Regards,
-Eric (he/him)

mhmodayman, 2023-04-16:
Hi Eric

I updated the functions as below

```
void calcEulerInLFrame()
{
   double CBL[3][3];
   MxMT(SC[0].B[0].CN, SC[0].CLN, CBL);

   C2A(321,CBL, &SC[0].Yaw_L, &SC[0].Pitch_L, &SC[0].Roll_L);
}
```

```
void calcEulerRatesInLFrame()
{   
   double CBL[3][3];
   MxMT(SC[0].B[0].CN, SC[0].CLN, CBL);

   MxV(CBL, SC[0].B[0].wn, SC[0].wbn_L);
   MT(SC[0].CLN, SC[0].CNL);
   MxV(SC[0].CNL, SC[0].wln, SC[0].wln_L);

   SC[0].wbl_L[0] = SC[0].wbn_L[0] - SC[0].wln_L[0];
   SC[0].wbl_L[1] = SC[0].wbn_L[1] - SC[0].wln_L[1];
   SC[0].wbl_L[2] = SC[0].wbn_L[2] - SC[0].wln_L[2];
}
```

The reason I wanted to have rates expressed in L frame, is that I wanted to see their initial values.

As you can see, in 42/Tx/SC_Simple.txt, I have angular rates wrt L frame as (0.08, -0.02, 0.03)
```
*************************** Initial Attitude ***************************
LAL                           ! Ang Vel wrt [NL], Att [QA] wrt [NLF]
0.08  -0.02   0.03            ! Ang Vel (deg/sec)
0.0    0.0    0.0    1.0      ! Quaternion
0.0    0.0    0.0    123      ! Angles (deg) & Euler Sequence
***************************  Dynamics Flags  ***************************
```

But, from socket I get different initial values.
```
SC[0].wbl_L = 1.124118660126e-03 -5.782522778693e-04 -1.424477398272e-04
```

Shouldn't values be like the following?
```
SC[0].wbl_L = 0.08 -0.02 0.03
```

ericstoneking, 2023-04-18:
Hi Mahmoud,

The input file takes in numbers in deg/sec.  The outputs are in rad/sec.  There's more going on, though.

Angular rates are the rate *of* some frame *with respect to* some frame.  So far so good.  But we have to store the components of that vector *expressed in* some frame.  If notation fatigue just set in, you're not alone.  The variable names I give call out the "of" and the "with respect to", but leave the "expressed in" implicit.  Sorry about any confusion that comes out of that.

B.wn is the angular rate of B with respect to N, expressed in B.
SC.wln is the angular rate of L with respect to N, expressed in N.

Look in the variable declarations in, for example, 42types.h, for hints about what frame a vector is expressed in.

Try:

void calcEulerRatesInLFrame()
{   
   double CBL[3][3];
   MxMT(SC[0].B[0].CN, SC[0].CLN, CBL);

   MTxV(CBL, SC[0].B[0].wn, SC[0].wbn_L);
   MxV(SC[0].CLN, SC[0].wln, SC[0].wln_L);

   SC[0].wbl_L[0] = SC[0].wbn_L[0] - SC[0].wln_L[0];
   SC[0].wbl_L[1] = SC[0].wbn_L[1] - SC[0].wln_L[1];
   SC[0].wbl_L[2] = SC[0].wbn_L[2] - SC[0].wln_L[2];
}

Note, however, that the numbers in the input file are meant to be expressed in the B[0] frame, not in the L (or N) frame.  So if you're checking the inputs, you probably want to express SC[0].wln in the SC[0].B[0] frame, and perform the subtraction there.

Regards,
-Eric (he/him)

mhmodayman, 2023-04-19:
Okay I made changes as you suggested

Here are the two updated functions

```
void calcEulerInBFrame()
{
   // SC[0].CLN:     DCM of L in N (or from N in L)
   // SC.B[0].CN:    DCM of B in N (or from N to B)
   // CLB:           DCM of L in B (or from B to L)

   double CLB[3][3];
   MxMT(SC[0].CLN, SC[0].B[0].CN, CLB);  // CLN * (CBN)^T = CLN * CNB = CLB

   // SC[0].A_B = [Roll_B, Pitch_B, Yaw_B]  [rad]
   // Roll_B:  euler angle (roll)  of B in L (expressed in B)
   // Pitch_B: euler angle (pitch) of B in L (expressed in B)
   // Yaw_B:   euler angle (yaw)   of B in L (expressed in B)

   C2A(321,CLB, &SC[0].A_B[2], &SC[0].A_B[1], &SC[0].A_B[0]);
}
```

```
void calcEulerRatesInBFrame()
{
   double CNB[3][3];
   double wln_B[3];

   // SC.B[0].CN:    DCM of B in N (or from N to B)
   // CNB:           DCM of N in B (or from B to N)

   MT(SC[0].B[0].CN, CNB);  // CBN^T = CNB

   MxV(CNB, SC[0].wln, wln_B);  // omega of L in N (expressed in N) --> omega of L in N (expressed in B)

   //SC[0].B[0].wn: omega B in N, (expressed in B) (aka) wbn_B

   SC[0].wbl_B[0] = SC[0].B[0].wn[0] - wln_B[0]; // wbn_B[0] - wln_B[0]  [rad/s]
   SC[0].wbl_B[1] = SC[0].B[0].wn[1] - wln_B[1]; // wbn_B[1] - wln_B[1]  [rad/s]
   SC[0].wbl_B[2] = SC[0].B[0].wn[2] - wln_B[2]; // wbn_B[2] - wln_B[2]  [rad/s]
}
```

And this is the Tx
```
TIME 2023-091-00:00:01.000000000
Orb[0].PosN = 6.978140907132e+06 4.226301118406e+03 6.265749079331e+03
Orb[0].VelN = -8.185734724099e+00 4.226299465848e+03 6.265746629312e+03
SC[0].A_B = -7.506400239794e-02 1.762589336543e-02 -2.597726773609e-02
SC[0].wbl_B = 7.022026052418e-02 -1.529406474823e-02 2.003880062415e-02
```

And this is the Inp_IPC.txt
```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
TX                                      ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
SERVER                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
TRUE                                    ! Allow Blocking (i.e. wait on RX)
TRUE                                    ! Echo to stdout
4                                       ! Number of TX prefixes
"Orb[0].PosN"				! Prefix 0
"Orb[0].VelN"				! Prefix 1
"SC[0].A_B"				! Prefix 2
"SC[0].wbl_B"				! Prefix 3
```

Now questions time :)

Euler angles are close enough to what I expected, but they are negated. Did I do something wrong?
I think that euler angles signs should follow the signs of the angular velocities at least for the initial values, right?
So it should be (+ve, -ve, +ve)
```
Expected values  =  0.07981115733 | -0.01969864003 |  0.029956
Actual Tx values = -0.07506400239 |  0.01762589336 | -0.02.597
```

For the angular velocities, I think they are now as expected, no need to change anything.
```
Expected values  = 0.08             | -0.02             | 0.03
Actual Tx values = 0.07022026052418 | -0.01529406474823 | 0.02003880062415
```

mhmodayman, 2023-04-25:
any thoughts @ericstoneking ?

closed 2023-04-28 by mhmodayman