ericstoneking/42 #117: Tx issue

mhmodayman, 2023-04-28:
Hello Eric

This issue is related to this one (https://github.com/ericstoneking/42/issues/116)

I am using your Tx functionality to simulate orbital dynamics (translational + rotational).

I noticed something strange that I want to ask about.

After simulating orbital dynamics using NASA 42, Princeton, and Basilisk simulators with a given set of initial conditions, it was observed that the angular rotation of the body frame around the orbit frame in the z-direction exhibited significant fluctuations when using the NASA 42 simulator. This was not the case with the Princeton and Basilisk simulators, where the angular rotation remained constant.

For all simulators, the initial conditions for the angular rotation were (0.08, -0.02, 0.03) in the x, y, and z directions, respectively.

The first value in the z direction obtained from NASA 42 simulator was:---------------- 0.02
The first value in the z direction obtained from the Princeton simulator was:----------- 0.03
The first value in the z direction obtained from the Basilisk simulator was:-------------- 0.03

**After 3 minutes...**
The value in the z direction obtained from NASA 42 simulator was:--------------------- -4.04e-04
The value in the z direction obtained from the Princeton simulator was:---------------- 0.03
The value in the z direction obtained from the Basilisk simulator was:------------------- 0.03

Screenshots from NASA 42 simulator are attached showing this incident.
Notice the variable SC[0].wbl_B at the beginning and after 3 minutes, representing the angular rate of the body frame about the orbit frame.
[image]
[image]

I have also attached Tx folder that I am using.
[Tx-20230428T085622Z-001.zip](https://github.com/ericstoneking/42/files/11351827/Tx-20230428T085622Z-001.zip)

I find this very strange, as there isn't any control applied.

**Is there any sort of control applied that makes the value goes down from 0.02 to -4.04e-04, while in Basilisk, for the same scenario value remains constant at 0.03?**

ericstoneking, 2023-04-28:
Hi Mahmoud,

There is control applied, because you have PROTOTYPE_FSW in your SC_Simple.txt (thanks for including your whole Tx directory).  Use PASSIVE_FSW instead and see what happens.  See Docs/FSW Models.pdf for more.

Regards,
-Eric (he/him)

mhmodayman, 2023-04-28:
Thanks, you're right

Could you please tell me also the range of euler angles (if they have range)?

are they limited in range like from (-2pi to 2pi) or no limitation applies?

ericstoneking, 2023-05-01:
The internet knows all.

mhmodayman, 2023-05-02:
I knew it by experiment

roll, pitch, yaw ranges from -2pi to 2pi

actually internet doesn't know all. :)

I think you misinterpreted my question, I meant to say if you are applying any limiters on the angles in your code (which is the case from my findings), correct me if I am wrong

For instance, I have a similar orbit propagation code, but without limiter
So you can find values for angles as such: 20.2, 30.5, -50.2  (and can be converted to angles within range from -2pi to 2pi easily).

You got what I mean?

closed 2023-05-09 by mhmodayman