ericstoneking/42 #118: guidelines for installing on Linux

capsulecorplab, 2023-05-02:
Hello! I'm attempting to install 42 from the zip file, [42 20230210.zip](https://sourceforge.net/projects/fortytwospacecraftsimulation/files/42%2020230210.zip/download) downloaded from sourceforge, but have been encountering a [non-zero return code](https://github.com/ODU-CGA-CubeSat/sealion-workspace-image/actions/runs/4855694909/jobs/8654504252#step:5:756) while running [make 42 && make install](https://github.com/ODU-CGA-CubeSat/sealion-workspace-image/pull/28/files#diff-d231edbab049480bb9012864178dcd0b9ccef7997ec46e14fc16ffe762b56f6cR178) for an Ubuntu 22.04 virtual environment. Any ideas as to what could be the problem?

ericstoneking, 2023-05-03:
I don't know what "make install" does, which means that you don't need it.  Try just "make" or "make 42" from the directory where the makefile is.  You should get a command-line executable in the same directory.

ericstoneking, 2023-05-04:
Hmm.  Your environment is more sophisticated than I am, so I don't know how much help I can be.  But if make is generating errors, it might be helpful to see what those errors are.  Does /tmp/42_build_output.txt have anything informative in it?

dmccomas, 2023-09-05:
I just installed and ran 42 on Windows WSL using Ubuntu 22.04 without any issues. The 5/4/23 logs in this issue have expired so I can't view them.  I've had this environment setup for C development prior to trying the 42 installation so I can't step you through a clean install. This is the environment I think you need:

sudo apt update
sudo apt install build-essential g++
sudo apt-get install freeglut3-dev
You may need these if they're not included with the freeglut3-dev package:
sudo apt install libgl-dev libglfw3-dev

A build error like "#include <GL/glut.h> not found" indicates a GL installation issue.  If you comment out "GUIFLAG = -D _ENABLE_GUI_" in the Makefile then you should be able to build and run 42 without graphics just to show that everything except the GUI is working correctly.  

To run 42 enter "./42"

You should also issue a "make clean" before you re-building after an erroneous build.

closed 2026-02-25 by ericstoneking