ericstoneking/42 #119: Issue with external control commands

mhmodayman, 2023-05-09:
Hello Eric

Hope I am not bothering you too much.

I am running the standalone version of the code, and trying to send wheel torque from a python client.

But I have issue with that, let me explain.

(Note I attached files I used from folder Standalone, so you can check if I am doing something wrong)

First, I run the standalone version
```
./42 Standalone
```
and I get the following on terminal
```
0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000
Server is listening on port 10001

Server side of socket established.
TIME 2020-001-00:00:00.000000000
Orb[0].PosN = 6.978137000000e+06 0.000000000000e+00 0.000000000000e+00
Orb[0].VelN = -0.000000000000e+00 4.226300000000e+03 6.265800000000e+03
SC[0].A_B = 0.000000000000e+00 -2.220446049250e-16 -5.551115123126e-17
SC[0].Q_B = -3.104126280195e-17 -1.015321020545e-16 1.387778780781e-17 1.000000000000e+00
SC[0].wbl_B = 8.060564245552e-02 -2.033867968489e-02 2.949788638083e-02
[EOF]
```

The issue starts when I send the following command from python
```
client_socket.send(b'SC[0].AC.Whl[0].Tcmd = 4[EOF]\n')
```

A message appears on the terminal as following,
Notice that the message is truncated. The part SC[0] of the message doesn't show up.
```
.AC.Whl[0].Tcmd = 4[EOF]
Imsg limit exceeded
```

However, 42 sends Ack to the python client, which I receive as following.
```
client_socket.recv(4)  # receive ack on sent the command
```

But the states values are not affected at all by the torque value that I sent. I am not sure what am I doing wrong.

I will put here sample of terminal printings for reference
```
0.0 SC[0] qrl = [0.0  0.0  0.0  1.0]
Reached CmdScript EOF at Time = 0.000000
Server is listening onnnnnnnnnnnnnnn port 10001
Server side of socket established.
TIME 2020-001-00:00:00.000000000
Orb[0].PosN = 6.978137000000e+06 0.000000000000e+00 0.000000000000e+00
Orb[0].VelN = -0.000000000000e+00 4.226300000000e+03 6.265800000000e+03
SC[0].A_B = 0.000000000000e+00 -2.220446049250e-16 -5.551115123126e-17
SC[0].Q_B = -3.104126280195e-17 -1.015321020545e-16 1.387778780781e-17 1.000000000000e+00
SC[0].wbl_B = 8.060564245552e-02 -2.033867968489e-02 2.949788638083e-02
[EOF]

.AC.Whl[0].Tcmd = 4[EOF]
Imsg limit exceeded
TIME 2020-001-00:00:01.000000000
Orb[0].PosN = 6.978132907123e+06 4.226299173718e+03 6.265798774976e+03
Orb[0].VelN = -8.185753492930e+00 4.226297521154e+03 6.265796324929e+03
SC[0].A_B = 7.949317114309e-02 -2.196952499847e-02 2.908859154588e-02
SC[0].Q_B = 3.988915194735e-02 -1.039682655457e-02 1.496785659586e-02 9.990378995931e-01
SC[0].wbl_B = 8.009468180521e-02 -2.190536305475e-02 2.946003961353e-02
[EOF]

.AC.Whl[0].Tcmd = 4[EOF]
Imsg limit exceeded
TIME 2020-001-00:00:02.000000000
Orb[0].PosN = 6.978120628496e+06 8.452593389744e+03 1.253159019981e+04
Orb[0].VelN = -1.637149738323e+01 4.226290084619e+03 6.265785299720e+03
SC[0].A_B = 1.579418852284e-01 -4.774252722580e-02 5.609277225298e-02
SC[0].Q_B = 7.950265896121e-02 -2.157362119133e-02 2.982960530757e-02 9.961546570357e-01
SC[0].wbl_B = 7.954933436929e-02 -2.346957885351e-02 2.942427327215e-02
[EOF]

.AC.Whl[0].Tcmd = 4[EOF]
Imsg limit exceeded
TIME 2020-001-00:00:03.000000000
Orb[0].PosN = 6.978100164134e+06 1.267887769040e+04 1.879736692438e+04
Orb[0].VelN = -2.455722206828e+01 4.226277690404e+03 6.265766924386e+03
SC[0].A_B = 2.353430810511e-01 -7.708767279910e-02 8.064395852597e-02
SC[0].Q_B = 1.187602361171e-01 -3.350770235514e-02 4.452281297311e-02 9.913581387799e-01
SC[0].wbl_B = 7.897043230676e-02 -2.503079574554e-02 2.939071318589e-02
[EOF]
```

Standalone folder attached below
[Standalone.zip](https://github.com/ericstoneking/42/files/11428423/Standalone.zip)

mhmodayman, 2023-05-14:
this is a gentle reminder @ericstoneking ))

ericstoneking, 2023-05-15:
Did you run the Standalone example successfully?  If not, then you should do that first, and if there are issues, maybe I can help you resolve them.

If you have run Standalone successfully, then the issue is with your code, and out of my scope.  It's your responsibility to find your way forward.  Happy coding!

mhmodayman, 2023-05-21:
Hello @ericstoneking 

Let me put the question in other form due to other observations from multiple tests.

I found that no matter what I send, I will get the warning "Imsg limit exceeded" if the length of the transmitted message is more than 5 characters

For example:
```
By sending the following, I get no warning 
socket.send(b'TIME')
But sending the following will show the "Imsg limit exceeded" warning
socket.send(b'TIME 2020-001-00:00:10.000000000')
```

Also I am not sure if I want to update the time, as an example, what format should I use from the following:
```
1- socket.send(b'TIME 2020-001-00:00:10.000000000')
2- socket.send(b'TIME 2020-001-00:00:10.000000000 \n')
3- socket.send(b'TIME 2020-001-00:00:10.000000000 [EOF] \n')
4- socket.send(b'TIME 2020-001-00:00:10.000000000 \n [EOF]')
or to separate every 4 characters by delimiter like
5- socket.send(b'TIME \n2020\n-001\n-00:\n00:1\n0.00\n0000\n000\n')
6- socket.send(b'TIME [EOF]2020[EOF]-001[EOF]-00:[EOF]00:1[EOF]0.00[EOF]0000[EOF]000[EOF]')
7- socket.send(b'TIME [EOF]2020[EOF]-001[EOF]-00:[EOF]00:1[EOF]0.00[EOF]0000[EOF]000[EOF]\n')
```

Could you help me with this issue, please?
I have been struggling with it since long

mhmodayman, 2023-05-29:
For anyone having the same issue, this solved it:

```
read_bytes = client_socket.recv(416)
client_socket.send(b'Ack \n')

client_socket.send(b'SC[0].AC.Whl[0].Tcmd = 1 \n[EOF]')
client_socket.recv(4)  # receive ack on sent command

read_bytes = client_socket.recv(416)
client_socket.send(b'Ack \n')

client_socket.send(b'SC[0].AC.Whl[1].Tcmd = 2 \n[EOF]')
client_socket.recv(4)  # receive ack on sent command
```

closed 2023-05-29 by mhmodayman