ericstoneking/42 #120: Orbit propagation consistency

f-e-d-e, 2023-05-19:
Hello Eric! 
I'm currently struggling to understand how to correctly propagate the orbit dynamics in 42 and I'd like your help in solving some doubts.
I'll report here the results for a 6U satellite (no sensor or actuator modeled, only a single body) in LEO using only the spherical harmonics gravitational field and no other perturbations. I've compared the results obtained with 42 with the ones given by GMAT in the same testing conditions (LEO orbit, simulation over 90 days using a 10s integration step, EGM96 gravity field model, max degree 2, RK4 numerical integrator).
I've done different tests in 42 considering different configurations for the input files and for the integration options, but none of them provides reasonable results in terms of orbital parameters profile. As I was mentioning, 42 simulation is pretty straightforward as I'm considering a simple satellite equipped with the PASSIVE_FSW. For each simulation test in 42, I retrieve the keplerian parameters from the output files PosN and VelN and I compare their profile with GMAT.

In the attached folder the following set of test results is available:
- **Test 0**: only the _Secular Orbit Drift due to J2_ in the orbit file is set to TRUE, while the _Gravity Perturbation Forces_ are FALSE. The results here seem acceptable, as only the secular drift is included (in the zip folder I included also the input files that I used to generate the results)
- **Test 1**: the _Secular Orbit Drift due to J2_ is now set to FALSE, while the _Gravity Perturbation Forces_ are TRUE, considering the Earth Gravity model with N=2 and M=2; the _Orbit Prop_ is set to _ENCKE_. As it can be seen in the plot, the parameters profile is quite strange if compared to GMAT
- **Test 2**: same perturbations as the previous test but with the _Orbit Prop_ is set to _COWELL_. Also here, the result is quite strange. While the eccentricity profile seems okay, the SMA and INC profiles are wrong

Here the zip folder containing the results:
[42_orbit_propagation.zip](https://github.com/ericstoneking/42/files/11518106/42_orbit_propagation.zip)

I've done a lot of other tests considering different inputs (by changing the _Dynamics Flags_ options, activating both the secular drift and the gravity perturbation forces) but none of them seem to provide reasonable outputs. Am I missing something? How should I propagate the orbital dynamics of a LEO satellite to take into account correctly the gravity perturbation forces?

Thanks
-Fede

ericstoneking, 2023-05-22:
Hi Fede,

Thanks for the data.  I see what you're talking about.  I'll have to squirrel this away till I have some time to devote to it.

In the meantime, the easiest way to get 42 to agree with GMAT is to use GMAT outputs as 42's inputs.  I've done this in the past for three-body orbits, and most of the code is in place for central orbits as well.  Looking at the code, it appears that there's one or two missing connections.  Search 42init.c for INP_SPLINE, and 42ephem.c around line 500, and you'll see what I mean.  I should be able to get this done within a week.

Regards,
-Eric

ericstoneking, 2023-05-26:
Hi Fede,

"Fixing the propagation core in 42" is a months-if-ever timeframe.  This is not my day job, and the only sure way to achieve 100% agreement with GMAT is to reverse-engineer it.  (Old saying: "He with one watch knows what time it is; he with two watches is never sure.")

Let me make a suggestion, though, that might be made to fit your needs.  The Orb_ input file defines the *Reference Orbit*.  The spacecraft can wander from its reference orbit under the influence of whatever forces aren't bookkept by the reference orbit.  So whatever GMAT can model, put those influences in the reference orbit, and apply the leftover forces (thrusters, solar sails, etc) to the S/C, with "Orb Prop" set to ENCKE.  This plays to the strengths of both GMAT and 42, and, if it fits your requirements, you're good to go today.

Regards,
-Eric (he/him)

mhmodayman, 2023-05-28:
Hello @ericstoneking  and @f-e-d-e 

@ericstoneking, First, I would like to thank you, you already helped me a lot with issues I encountered while using your code.

After reading your discussion, I am curious to confirm my understanding that orbit propagation is not accurate in 42.

@ericstoneking, I am using your code as part of my work, so it is very important for me to have correct results.

Regards
Mahmoud

f-e-d-e, 2023-06-09:
Hi Eric,

 I followed your suggestion and I'm using GMAT splines in the Orb file to initialize the reference orbit. 
I include the main orbital perturbations inside GMAT (gravity, drag, ...) while I use 42 to add forces to simulate the thrust effects. However, something does not work as expected.
I use the PROTOTYPE_FSW to maintain the attitude aligned with the velocity vector of a 6U S/C and I implemented a second controller to apply an ideal force in the S/C CoM (through the IdealFrc) to simulate a constant low-thrust force.
By performing a simple simulation with a reference orbit coming from GMAT that only includes J2 perturbations and a constant force (of 2.5mN) applied direclty in 42 and fixed on the velocity vector, the result shows the strange behavior reported below.
Here is the simple propagation obtained from GMAT (J2 perturbation) without any force added in 42
[image]
Here, instead, is the simulation with the constant force applied
[image]

As it is possible to notice from the simple propagation, the reference orbit is loaded correctly from 42, but the problem arises when I try to include an external force in the model.
Do you have any suggestions on what I am missing?

Thanks,
-Fede

ericstoneking, 2023-06-09:
One thing to check: are you exerting your Ideal Actuator force in B[0] or N?  Looking at 42actuators.c, it's expected to be expressed in B[0].

Beyond that, you might be on your own.  I threw together a quick example on my installation, and I get plausible results.  But I'm not pushing it as hard as you are, so your mileage may vary.