ericstoneking/42 #122: How to give different initial angular velocities of each body? and...

caution-lee, 2023-06-22:
Hello, Eric!

I'm Joohee, a GN&C Engineer who use 42 very useful!

I have a few questions, so I'am writing like this.

Now, I'm participating in Active Debris Remove mission.
I want to implement the movement after the satellite captures space debris using a net through 42.

I'm currently thinking to implement the multi-body dynamics model of capturing space debris using Tether-net from a satellite.

**Implement satellite, Tether-net, and space debris inside one S/C in 42**
-> This is because I couldn't find a way to physically connect Satellite and Debris composed of each S/C.

**Question!!**
Q1. After implementing the three S/Cs (ADR Satellite, Tether-net system and Space debris), is there a way to physically connect the Satellite (S/C 1) and Space debris (S/C 3) (e.g. Tether-net in my system) to check the interaction of the motion?
(The implementation of the tether system will use sub-mass that implements numerous bodies.)

Q2. I want to give the ADR satellite and Space debirs different initial angular velocities. but, (according to what I have studied your code), in the code of 42, the angular velocity information entered in the SC_XX.txt file is set in the main body.
Is there a way to give different initial angular velocities of each body in a system implemented with one S/C?

(It is impossible to set each angular velocity, so I'm trying to implement a method of connecting three S/Cs (Satellite, Tether-Net, Space debris))

I look forward to hearing from you.

Regards,
Joohee

ericstoneking, 2023-06-23:
Hi Joohee,

One approach is to model your system as one SC with multiple bodies, and put whatever complicated force systems you need into the joint models.

For discussion, consider one two-body SC.  It will have one joint (42 only supports tree topologies).  Look at Demo/SC_TumbleSat.txt for an example of how to set up a passive joint.  Note that a PASSIVE joint with 3 rotational DOF and 3 translational DOF and zeros for spring and damping constants is one way to model a free-floating body.  With a little custom coding, you should be able to change the joint parameters as you like, even during the course of a sim run.  (See 42cmd.c:SimCmdInterpreter for inspiration.  See Include/42types.h for the variables you want to alter.  They'll be in the JointType struct.)  Maybe even better, make the joint type AD_HOC_JOINT, and customize it in 42joints.c:AdHocJoint.

To set the angular rates of an outer body, use the joint rates.  These are the rates of the outer body relative to the inner body.  So if you want your ADR satellite starting with a 5 deg/sec clockwise rate, and your debris with a 5 deg/sec counterclockwise rate, make the joint rate be 10 deg/sec counterclockwise.

Hope this helps.

Regards,
-Eric

caution-lee, 2023-07-28:
Eric,

Thank you for your response.

I given angular velocity to debris on your advise, finally success! 👍
So, I'm completed an ADR mission system consisting of 5 submass tether, ADR satellite and debris moving at 5 deg/sec relative to ADR satellite in one spacecraft.

Unfortunately, I am having difficulty giving "Spring/Damping Coefficient" value in Joint parameter.

The spring/damping factor must be calculated using the material properties of nylon and assigned to the tether.
(Because, in real experiments, we use a tether made of nylon.)
I used Young's modulus and Hooke's law to calculate the Translational Spring coefficient, but it is very confusing because it has a value of 10e5 level.

Used material/physical properties are below
=====================================
Young's Modulus (E) = 2.4 Gpa
Tether length (L) = 3 m
Tether radius = 0.005m
Tether Cross-section area (A) = 0.005*0.005*pi
Stiffness (K) = EA/L = 6.2832e+04 N/s
=====================================

[Spacecraft setting]
ADR satellite = 80 kg
Debris = 30kg
Tether = 0.2170 kg (Total mass, 0.0434 kg each sub-mass)

**So, I am wondering how to calculate (or determine) Spring/Damping coefficients.**

+)
Currently, the tether is modeled in the form of a slender rod, and the moment of inertia (MOI) value is given very small (almost zero) for the Y axis.
When setting the simulation in this way, the simulation is stopped due to the Gimbal Lock.
(Since the MOI of the Y-axis is very small, it seems to show a rapid rotational/translational motion change even with small disturbances.)

**I separated the 3m Tether into 5 submass and modeled it as a slender rod and entered the value, I wonder if this is the right simulation.**

Thank you for your quick response and continuous updates all the time.

Regards,
Joohee

ericstoneking, 2023-08-01:
Hi Joohee,

You're on the right track.  A couple of suggestions:
1)  I suspect that your simulation timestep is too large.  Study up on how to properly set it.
2)  Try making your tether joints be spherical instead of gimballed.

Regards,
-Eric (he/him)