ericstoneking/42 #123: Proximity operations simulation

john-nyy, 2023-07-10:
Hi Eric,
 thank you very much for all your efforts in developing and maintaining this repo. I find it really useful.
I've some doubts on how to properly set up a simulation with two spacecrafts - a chaser and a target - that are in a near-range proximity. I would like to start from a simple scenario where the relative motion is a free motion simply described through CW equations (EULER-HILL propagation mode in 42 I assume) and then add the different perturbations.
To do so, I thought that a good way was to use the formation flying options. As of now, I've created the followings:

- a single LEO reference orbit that is located at the origin of the formation
- the target S/C, located in the origin of F and propagated with the FIXED orbit propagation to be exactly on the reference orbit
- the chaser spacecraft that has an offset in its CM position wrt F to setup the initial relative distance with the target. I'm using EULER_HILL to propagate the chaser, with no perturbation effect active

I added to the report.c two output files (PosEH.42 and VelEH.42) to track the position of the chaser wrt the target (that is also the origin of F). However, the output files do not seem reasonable. Am I doing something wrong?
I know that the question is not an issue but more of a help request, but I really don't know how to setup a correct proximity operations scenario and I didn't find similar simulations in the documentation or in the Demo folder.

Thanks

ericstoneking, 2023-07-10:
Hi Johnny,

Take a look at Docs/Flight Regimes.pdf.  Also, there are a few examples of prox ops in the Demo.  Look at SC_BBM, SC_Moai, SC_ISS, SC_Shuttle, SC_ThrDemo, and all three SC_Cubesats.  Select "Prox Ops" from the "Show" menu in the "42 Cam" window to see some coordinate grids.  The brownish curve is the natural EH motion projected forward and backward in time.

Yes, CW = Euler-Hill.

EH might indeed be broken.  I recall fiddling with orbit stuff several months ago at someone's request, and may not have fully tidied up after.  You can help diagnose by describing what you did, what you expected to see, and what you actually saw.

Regards,
-Eric (he/him)

john-nyy, 2023-07-10:
Hi Eric,
 thanks for your answer. I'll paste here the tests that I'm currently doing but, as I was saying, there may be something wrong in how I'm setting up the simulation scenario.

Below there is a .zip folder containing the main input files that I'm using. I took the input files inside the Demo folder as the main reference and I'm using:

- a single reference orbit (Orb_Cube.txt) that is a circular LEO orbit (400km)
- two S/C are attached to the same reference orbit, a Chaser (SC_0) and a Target (SC_1). I used the same input files that is used in the Demo folder, considering 6U cubesats for both of them. Both spacecraft have uncontrolled attitude (PASSIVE_FSW) since I'm only interested in the CoM relative motion. Also, the initial attitude in aligned with the L frame and a null angular speed in the L frame is initialized. Both the spacecraft are really simple and consist in just a single body without any sensor/actuator
- the Chaser starts with a CM offset of X=15m wrt F while all the others offsets are 0. The propagation method is set to ENCKE (I've tried also with EULER_HILL but the results show the same profiles)
- the Target is placed in the origin of F and a FIXED propagation is used to maintain its alignment with the reference orbit
[input_files.zip](https://github.com/ericstoneking/42/files/12003248/input_files.zip)

I've done some tests, adding to the report files both PosEH.42, VelEH.42 and PosF.42, VelF.42. Their profiles are comparable. Here I attach the results considering a first scenario where no perturbation is used (in Inp_Sim.txt all the flags are FALSE) and a second one where I activated the Gravity Perturbation forces with a 2,2 degree for the gravity model. In both cases, the results are pretty strange and are not comparable with the standard CW equations (I've done the same tests by running a script that retrieves the CW positions in time).
[image]
[image]

As I was saying, the problems may be in the simulation scenario. I tried to replicate what I saw in the Demo folder but I may have lost something.

Thanks

ericstoneking, 2023-07-12:
Hi Johnny,

Yes, my L frame is +Z to nadir, +Y opposite the orbit normal, and +X completing the right-handed triad.

So your other source and 42 agree.  I still have a nagging question about setting up initial conditions, Encke, and my numerical integrator, but that's more of a philosophical issue, if the results you show are of adequate accuracy for your purposes.

Moving on to adding geopotential harmonics.  I agree that the amount of growth is startling, but what's your basis for concluding that it's wrong?

Here's an experiment to try:  

1)  Set up your chaser as your only S/C for the time being.  Perform a run with FIXED orbit propagation.  42 will write out PosN.42 and VelN.42 (for SC[0] only, so we want the chaser to be SC[0] for this experiment).  Save those off with unique file names.  

2)  Now, in your Orb_ file, set "Secular Orbit Drift Due to J2" to true.  Run the sim again, and save PosN.42 and VelN.42 with unique file names.

3)  Set the Secular Drift flag back to false.  In Inp_Sim, set Earth Gravity Model N and M to (2,0), and set Gravity Perturbation Forces to true.  In SC_Chaser.txt, set Orbit Prop to ENCKE.  Run the sim again, and save PosN and VelN.

4)  Compare the three sets of PosN and VelN.  What I see is that the results from steps 2 and 3 are in decent agreement, and noticeably different from the pure Keplerian motion of step 1.

My source for the secular drift due to J2 is Markley and Crassidis, "Fundamentals of Spacecraft Attitude Determination and Control", Appendix C.  I'm sure there are other references out there as well.

As a sanity check, you might look at the geopotential terms, and work out how much "extra" gravitational pull J2 exerts.  What would that do to the mean motion and semi-major axis of an orbit that was initialized assuming only Keplerian motion?

If you browse other issues in this repo, you'll find cases of 42 not agreeing with GMAT (over long time scales, especially).  So I'm not saying that 42 is as accurate as it can be with regard to orbit propagation; there is work to do (which I'll get around to someday, maybe).  It's your call whether or not it meets your needs as it stands.  If not, I recommend GMAT for hifi orbit propagation with gravitational perturbations.  You can use 42 to use that as a reference orbit, and put Encke on top of that to handle whatever perturbations GMAT can't handle gracefully.

Regards,
-Eric (he/him)