ericstoneking/42 #124: Isolating the graphics engine

clholtz73, 2023-07-11:
Hey Eric.  Great product!  I've been working space simulation a long time and very impressed with what you have with 42, especially for the size of the code base.  I'm actually interested in the possibility of extracting the 3d graphics code for use with other space simulations.  Does that seem doable? Any thoughts, tips, or docs that would be helpful?

ericstoneking, 2023-07-12:
Hi Corbin, 

Thanks for the kind words.

Doable, yes.  There will likely be labor in hooking up all the globals that the GUI expects.  All the graphics-specific code is segregated in a few source and header files, but liberal use is made of non-graphics-specific globals, so the interface isn't what you might call modular :-/

I'll throw another idea at you.  There are "game engines" out there that I've considered using as a graphics front end for 42 (there was a scare a few years back that OpenGL might become hard to sustain).  I looked at Unity, Lumberyard, Godot, and (I think) Unreal.  I settled on Godot for a few reasons (free open source, right-handed coordinates, expected support for VR).  OpenGL is still here, so I haven't pursued it further.  Depending on your needs and preferences, it might be just as easy for you to build a front end with these high-quality graphics tools as it would be to make all the connections to 42's graphics front end.

If you go that route, and don't mind sharing, keep in touch.

Regards,
-Eric (he/him)

clholtz73, 2023-07-12:
Thanks for the quick response, Eric.  The game engine path is possible, but adds additional dependencies and my previous look at them showed some issues with planetary distances & such.  We do have much higher fidelity capabilities, but they are difficult to deploy on a developer's desktop for daily use.  42 is so compact that having everyone popup the graphics when needed would be very lightweight.

I tried to identify the graphics code previously, but nothing really obvious popped out.  If you can point me at the appropriate files, that would be much appreciated.

Thanks!