ericstoneking/42 #127: JDtoGPSTime is not correct?

amilagili, 2023-12-20:
https://github.com/ericstoneking/42/blob/d4547dae44270876657aec009fe59980082ed999/Kit/Source/timekit.c#L296

Hi!

It seems like the function takes in TT time to calculate GPS Time, but the output time is 51.184 seconds ahead ?
Since TT time = GPS Time + 19S + 32.184S, should this be subtracted?

Thanks a lot

ericstoneking, 2024-01-02:
Yes, you're right.  See recent commit.  Let me know if I flubbed the sign.

amilagili, 2024-01-12:
Hi,

Thanks for your commit. 
The sign is ok, but it seems there's a loss in precision as a result of the calculation.

Is it correct to think that the JDToGpsTime() function is intended to calculate "the time spent since the GPS epoch(=GPS time) in JD (with the GPS scale), but it is not intended to convert TT to GPS time scale.

Hence, it may be incorrect to pass the TT time (TT.JulDay) to JDToGpsTime(), but rather the GPS time in JD should be passed instead?

For example, I set the simulation start time as 2022-10-30 00:00:00.000, but from the GPS what I read is 2022-10-30 00:00:00.000010~

ericstoneking, 2024-01-20:
That's roundoff error.  I've reduced it by going directly from AtomicTime to GpsTime (which is sec since J2000) to GpsDate (GpsRollover,GpsWeek,GpsSecond).  Now the roundoff is about 50 nanoseconds on my machine for dates in the 2020's.  If you need better than that, you may want to split GpsSecond into seconds and subseconds.