ericstoneking/42 #130: Question about paths and the length of the char arrays for paths

TenguBoss, 2024-02-29:
Aloha 42 Team.

I just installed `42` on WSL2 and I verified that it works correctly with the demos provided. It has improved a lot since I tried it about 10(?) years ago. Thank you!

When I tried to run it on a Windows directory I got an error that made me "reverse engineer" the following command line

`$ ./42 ../../../mnt/c/very/very/very/very/very/very/very/very/long/path/here/Missions/Agena-D_ADR/42sim/InOut`

The command failed with

`Error opening ssions/Agena-D_ADR/42sim/InOut/42.mtl`

I copied `42.mtl` to the directory, reran, and got the same error. I assumed that this had to do with path definitions so I edited `42.h` to increase the length of the path char arrays.

I changed 

`EXTERN char InOutPath[80];`
`EXTERN char ModelPath[80];`
`EXTERN char CmdFileName[80];`

to 

`EXTERN char InOutPath[512];`
`EXTERN char ModelPath[512];`
`EXTERN char CmdFileName[512];`

remade 42, ran the command line, and confirmed that it worked.

Now I have two questions:

1. Is my assumption about the string for the path, as seen in the command above, correct? (I haven't been able to find the function that assigns `ModelPath` and it seems that `42` assumes some form of relative path to generate `ModelPath`.)
2.  Is there a chance that I will run into problems because the path char arrays are now six times longer? If yes, should I find another workaround?

Before you suggest it, I am required to store the working (`InOut`) files in the Windows file system in specific directories.

Thank you.

BU

ericstoneking, 2024-02-29:
1)  Yes.  FWIW, ModelPath is set in 42init.c:InitSim.  The default is "./Model/", but you can replace that with the second argument in your command-line invocation of 42, as in "42 MyInOut MyModels".

2) Not that I'm aware of.  But then, I didn't have the imagination to encompass somebody needing a 512-character path in the first place :-).  There are several places where ModelPath is passed as an argument to a function.  I *think* that all of those have been made to accept "const char *Path" instead of the old "char Path[80]" style.  If you find any instances of the old way, let me know and I'll fix them.

TenguBoss, 2024-02-29:
Thank you Eric for the prompt reply. This was an edge case due to WSL2 + Windows requirements.  

I looked for `char Path[80]` with
 
`$ find ~/42/ -name "*.c" -exec grep -Hi "char [Pp]ath\[80\]" {} \;`

and found it here:

`/home/homedir/42/Kit/Source/geomkit.c:void WriteGeomToObjFile(struct MatlType *Matl,struct GeomType *Geom,const char Path[80]`
`/home/homedir/42/Kit/Source/glkit.c:GLuint PpmToRingTexTag(const char path[80], const char filename[80])`

As I was looking at the source code, I also noticed that `path[40]` came up in several places in source code in the `World` and `Kit` directories. 

As I mentioned in the previous email, I didn't get an error yet, but I worked on the simplest of the demos. I'll post an update if I encounter problems.

Best regards,

BU