ericstoneking/42 #132: Propellant slosh simulation in 42

TenguBoss, 2024-03-11:
I read in the 42 docs about simulation of propellant slosh but I cannot find the specific slide deck anymore.  I wrote in my research notes a question to find out if the specific CFD solver you mentioned employs Smoothed-Particle Hydrodynamics (SPH) or Volume of Fluid (VoF) methods. However, I didn't write down the CFD solver so I have to go back and search for it.

Before I go to down the rabbit hole of CFD solvers I would like to ask the 42 community if anyone has implemented a propellant slosh simulator that runs on-line within 42. If yes, does your slosh model run sufficiently fast to perform real-time, or even faster than real-time simulations?

In the meantime I have started a poor man's SPH-like simulator with MuJoCo. I only have a low-fidelity open cavity sim in one gee, see the rendering, but if looks interesting.

Thank you.
BU
[Geyser.pdf](https://github.com/ericstoneking/42/files/14563456/Geyser.pdf)

ericstoneking, 2024-03-12:
The commercial CFD solver we hooked 42 up to was Star-CCM.  It was decidedly slower than real time.

A popular way to model slosh is as a pendulum.  This is easy to do in 42, modeling the pendulum as an internal appendage body, with any spring and damping placed in the joint.

I, among many others, am looking into SPH as a way to fill the chasm between the pendulum models and the full-up CFD.  I'd like to pick the balance between speed and fidelity depending on where in the life cycle I am.  I'll get it worked out eventually, but you shouldn't wait for me.

Regards,
-Eric

TenguBoss, 2024-03-12:
Aloha @ericstoneking . Thank you for the prompt reply. 

I am familiar with the damped oscillator models of slosh. The application is Rendezvous and Proximity Operations (RPO) and I am interested in axisymmetric and asymmetric geyser-like propellant motion that it is know to occur even after a propellant settling maneuver. I don't think those dynamics are easily modeled by oscillators. 

For example, Mason and Starin published a paper on the geyser driven instabilities of the Solar Dynamics Observatory - not an RPO mission but a very good example of the intricacies of geyser driven instabilities. Mason, P. and S. Starin (2011). _The Effects of Propellant Slosh Dynamics on the Solar Dynamics Observatory_. AIAA Guidance, Navigation, and Control Conference, Portland, OR. 

I'll keep at it and post updates when I have something worthwhile.

Regards,
BU